stm32-rs/stm32f4xx-hal #484: No implementation for read/write multiple bytes for SDIO

OBRATEN, 2022-04-19:
For fat32 crate I need read/write multiple bytes functions, but there are not any, tried to implement them by myself, but lines of code are not public. Asking for implementation for those functions in HAL.

burrbull, 2022-04-19:
Main SDIO related code is in https://github.com/jkristell/sdio-host . Here is only wrapper. Have you asked there?

jkristell, 2022-04-20:
You can read and write in multiples of "blocks" of 512 bytes with:

```
pub fn read_block(&mut self, blockaddr: u32, block: &mut [u8; 512]) -> Result<(), Error>;

pub fn write_block(&mut self, blockaddr: u32, block: &[u8; 512]) -> Result<(), Error>;
```

mmmasteri, 2022-04-20:
> You can read and write in multiples of "blocks" of 512 bytes with:
> 
> ```
> pub fn read_block(&mut self, blockaddr: u32, block: &mut [u8; 512]) -> Result<(), Error>;
> 
> pub fn write_block(&mut self, blockaddr: u32, block: &[u8; 512]) -> Result<(), Error>;
> ```

There are command for multiple bytes listed in library, but in HAL there are no implementation for it, how do i read/write multiple bytes with this commands? Call them multiple times?

jkristell, 2022-04-20:
I'm not sure what you mean with no implementation? And yes you call them multiple times and with the blockaddress you want to read/write. 

Example:

https://github.com/jkristell/feather-f405/blob/master/examples/sd-usb-mass-storage.rs

mmmasteri, 2022-04-20:
> I'm not sure what you mean with no implementation? And yes you call them multiple times and with the blockaddress you want to read/write.
> 
> Example:
> 
> https://github.com/jkristell/feather-f405/blob/master/examples/sd-usb-mass-storage.rs

Thanks for example, im talking about functions from SDIO, that showed in sd example