stm32-rs/stm32f4xx-hal #492: Pin state violation in public API.

YruamaLairba, 2022-05-01:
Hi, i discovered several `Pins` trait allowing to violate the Pin type state system from outside the crate. It concern i2c, i2s, serial and spi module. The way to do this violation is actually simple.

Pseudo example
```rust
use stm32f4xx_hal::serial::Pins;

let pins = (gpioa.pa9, gpioa.pa10); // tuple of Input pins
pins.set_alt_mode(); // type state violation, set pin in alternate mode without changing the type.
```

A simple fix consist to put those trait in a private module to prevent using their methods outside the crate. The trait `SetAlternate` must also documented, since the dangerous pin state violation ability comes from here.

burrbull, 2022-05-02:
`SetAlternate` is in private `alt` mode

burrbull, 2022-05-05:
Yes, this is really not good.

burrbull, 2023-05-02:
Closing as API was changed