stm32-rs/stm32f4xx-hal #498: MCO configuration is missed

andrey-ilinykh, 2022-06-25:
I don't see a way to configure MCO (master clock output). Should it be be a part of Rcc?

YruamaLairba, 2022-06-27:
I looked, it seems it's just unimplemented yet, otherwise it would exist [here](https://docs.rs/stm32f4xx-hal/0.13.2/stm32f4xx_hal/rcc/struct.CFGR.html).

dvdsk, 2023-05-24:
Configuring the MCO only makes sense once the clocks have been configured right? Would it make more sense for it to live [here](https://docs.rs/stm32f4xx-hal/latest/stm32f4xx_hal/rcc/struct.Clocks.html)? 

I would not mind implementing it since me misconfiguring MCO using the pac caused me quite a lot of trouble recently: https://github.com/stm32-rs/stm32f4xx-hal/issues/643 .