stm32-rs/stm32f4xx-hal #503: Allow for changing serial baud rate after creation

iMobs, 2022-07-08:
My use case is to allow devices to negotiate a higher baud rate from an initial default. Currently, the speed gets set when creating a new Serial (or Rx/Tx). There isn't another way to change the speed afterward besides releasing and recreating the connection or manually changing the baud rate register. Would it make sense to move the setting of the baud rate from the new function to a separate public function that can be called on its own?

iMobs, 2022-07-08:
Another more esoteric use is changing the baud rate to send the 100us break before data of [DMX512](https://en.wikipedia.org/wiki/DMX512#Protocol).

burrbull, 2022-07-19:
It would be nice if you make PR. See [f1xx-hal](https://github.com/stm32-rs/stm32f1xx-hal/blob/f9b24f4d9bac7fc3c93764bd295125800944f53b/src/serial.rs#L409) for example