stm32-rs/stm32f4xx-hal #522: Help needed with understanding the Lifetime checker for USB Interrupts

Slamy, 2022-08-27:
It seems I'm unable to reopen #411 so I guess I at least try this again

I'm pretty new to embedded rust and tried to change the usb_serial.rs example to utilize the OTG_FS interrupt. Today is now the third day I'm trying to do this and I keep failing even after reading a lot about the life time checker. Curious ones might be able to read my [mess](https://github.com/Slamy/stm32f4xx-hal/tree/stm32_usb_irq).
As far as I understand, there seem to be two mutual exclusive solutions to IRQs for embedded cortex-m projects. With rtic and without it. I've now seen the example posted here and this seems to be a viable solution. Nonetheless I feel like this doesn't help me and others to fully understand the actual process behind this and being able to understand the internals of Rust in case we stumble upon future problems. The example blinky-timer-irq.rs demonstrates how it can be done as well. But it seems that the USB components are much more difficult to satisfy here. Would it be possible to provide an example based on blinky-timer-irq.rs and usb_serial.rs to show how it is done correctly without relying on rtic?

burrbull, 2022-08-27:
Without RTIC you need to work with NVIC manually. Also you need to wrap your data in `Mutex`es or other concurrency primitive to prevent data race.
https://github.com/stm32-rs/stm32f1xx-hal/blob/master/examples/blinky_timer_irq.rs#L35

Slamy, 2022-08-27:
I would like to thank you for this very fast response. As mentioned in my question, I'm already fully aware of the example. Please allow me to get into more detail.

According to your mentioned example of `blink_timer_irq.rs`, first I need something like this:

```rs
static G_USB_BUS: Mutex<RefCell<Option<UsbBusAllocator<UsbBus<USB>>>>> = Mutex::new(
    RefCell::new(None)
);
static G_USB_SERIAL: Mutex<RefCell<Option<usbd_serial::SerialPort<UsbBus<USB>>>>> = Mutex::new(
    RefCell::new(None)
);
static G_USB_DEVICE: Mutex<RefCell<Option<UsbDevice<UsbBus<USB>>>>> = Mutex::new(
    RefCell::new(None)
);
```

Those containers would allow me to safely provide those to the interrupt handler. Exactly like the blinker example does.
And now the problem starts.
```rs
    cortex_m::interrupt::free(|cs| {
        *G_USB_BUS.borrow(cs).borrow_mut() = Some(UsbBus::new(usb, unsafe { &mut EP_MEMORY }));

        *G_USB_SERIAL.borrow(cs).borrow_mut() = Some(
            usbd_serial::SerialPort::new(G_USB_BUS.borrow(cs).borrow().as_ref().unwrap())
        );
    });
```
This code doesn't compile and this is the wall I'm currently hitting. As opposed to the other example, this one here requires a reference to another thing. `UsbDeviceBuilder` and `usbd_serial::SerialPort` both require to have a reference to `UsbBusAllocator` and my current skill in Rust doesn't help me solve this.

The error message I'm getting here is:
```
error[E0521]: borrowed data escapes outside of closure
  --> src/main.rs:50:42
   |
46 |     cortex_m::interrupt::free(|cs| {
   |                                --
   |                                |
   |                                `cs` is a reference that is only valid in the closure body
   |                                has type `&'1 CriticalSection`
...
50 |             usbd_serial::SerialPort::new(G_USB_BUS.borrow(cs).borrow().as_ref().unwrap())
   |                                          ^^^^^^^^^^^^^^^^^^^^
   |                                          |
   |                                          `cs` escapes the closure body here
   |                                          argument requires that `'1` must outlive `'static`

error[E0716]: temporary value dropped while borrowed
  --> src/main.rs:50:42
   |
49 |         *G_USB_SERIAL.borrow(cs).borrow_mut() = Some(
   |          ----------------------- argument requires that borrow lasts for `'static`
50 |             usbd_serial::SerialPort::new(G_USB_BUS.borrow(cs).borrow().as_ref().unwrap())
   |                                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ creates a temporary which is freed while still in use
51 |         );
   |          - temporary value is freed at the end of this statement
```

There seems to be something with the `Mutex` which is special to the cortex_m crate. It would be superb if you may provide me with a hint here.

Thx

burrbull, 2022-08-27:
I can't give you hint as I never use this way. But you could ask your question in matrix room ([main](https://matrix.to/#/#rust-embedded:matrix.org) or [rtic related](https://matrix.to/#/#rtic:matrix.org)).

Last days new `critical-section` crate was released. Possibly it works better, I don't know.

My thought that you incorrectly choice structures which should be shared.

Technically you can don't use critical-section at all if you've not enabled interrupts yet.

Slamy, 2022-08-28:
I was able to fix my problem but I'm surely not amused.
This is evil:
```rust
let usb_dev = G_USB_DEVICE.borrow(cs).borrow_mut().as_mut().unwrap();
```
This is not
```rust
let mut usb_dev_refmut = G_USB_DEVICE.borrow(cs).borrow_mut();
let usb_dev = usb_dev_refmut.as_mut().unwrap();
```
It seems that the result of `borrow_mut()` must be stored in its separate variable to it is not immediately thrown away.

The other thing was about static lifetimes. I wasn't aware that `#[entry]` allows me to create safe `static mut` at the start of the function. Coming from other languages I find this concept difficult to understand. It's almost like the basic rules of the language are getting rewritten. This of course is not meant for you here as this project is about the HAL and not cortex_m_rt which is the culprit here. Nonetheless I feel lost about this.

Anyway... I was able to have now a running example with rtic and without it. This makes me happy.
While the example folder of this project might not be the right place for the rtic code, the other one might be.
It is basically usb_serial with interrupts. Shall I provide a PR to add it to the examples?

closed 2022-08-28 by Slamy

burrbull, 2022-08-28:
> Shall I provide a PR to add it to the examples?

Sure. Any contribution is welcome.