stm32-rs/stm32f4xx-hal #537: Writing to GPIOC

xiugaze, 2022-10-10:
Hi, 
I can't seem to figure out how to use the GPIOC pins outside of writing to each pin individually. I am trying to manually access the BSRR and the ODR/IDR, but those only seem to be implemented for GPIOA, B, and K. Is this the case or am I just missing something?

adamgreig, 2022-10-10:
If you're trying to access the registers directly then I guess you're using the PAC?

It's a separate crate ([stm32f4](https://crates.io/crates/stm32f4)) so it might be easier to check its documentation directly, but I guess what you're running in to is that the PAC has **modules** for [gpioa](https://docs.rs/stm32f4xx-hal/latest/stm32f4xx_hal/pac/gpioa/index.html), [gpiob](https://docs.rs/stm32f4xx-hal/latest/stm32f4xx_hal/pac/gpiob/index.html), and [gpiok](https://docs.rs/stm32f4xx-hal/latest/stm32f4xx_hal/pac/gpiokb/index.html), which contain a `RegisterBlock` and all the register types, but it has **structs** for [GPIOA](https://docs.rs/stm32f4xx-hal/latest/stm32f4xx_hal/pac/struct.GPIOA.html), [GPIOB](https://docs.rs/stm32f4xx-hal/latest/stm32f4xx_hal/pac/struct.GPIOB.html), [GPIOC](https://docs.rs/stm32f4xx-hal/latest/stm32f4xx_hal/pac/struct.GPIOC.html), D, E, F, G, H, I, J, K. It's just that most of them are identical, so C...K all share the same `RegisterBlock` type and registers.

So, you should find that the [`Peripherals` struct](https://docs.rs/stm32f4xx-hal/latest/stm32f4xx_hal/pac/struct.Peripherals.html) you get from [`take()`](https://docs.rs/stm32f4xx-hal/latest/stm32f4xx_hal/pac/struct.Peripherals.html#method.take) does contain a [`GPIOC` member](https://docs.rs/stm32f4xx-hal/latest/stm32f4xx_hal/pac/struct.Peripherals.html#structfield.GPIOC), and [its documentation](https://docs.rs/stm32f4xx-hal/latest/stm32f4xx_hal/pac/struct.GPIOC.html) points to the [`gpiok::RegisterBlock`](https://docs.rs/stm32f4xx-hal/latest/stm32f4xx_hal/pac/gpiok/struct.RegisterBlock.html) type, which is what contains those registers.

```rust
let dp = stm32f4xx_hal::pac::Peripherals::take().unwrap();
let gpioc = dp.GPIOC;
gpioc.bsrr.write(|w| w.bs3().set());
```

xiugaze, 2022-10-11:
Ok, that makes sense. In the line
```
gpioc.bsrr.write(|w| w.bs3().set());
```
what gets passed as the argument to the closure? Is `w` just the value that I want to write?

burrbull, 2022-10-11:
https://docs.rs/svd2rust/latest/svd2rust/#read--modify--write-api

xiugaze, 2022-10-12:
Ok, understood.

closed 2022-10-12 by xiugaze