stm32-rs/stm32f4xx-hal #549: Implement complemntary PWM and dead-time generation

janschiefer, 2022-11-06:
STM32F4xx and STM32F1xx devices are optimal for driving mosfets in hals or full bridge configuration.

Implement complemntary PWM and dead-time generation should be implemented.

janschiefer, 2023-01-17:
Great!

I'll test it in about two weeks, since currently I have no access to my development system.

janschiefer, 2023-02-14:
Well... how do i use it?

`    let gpioa = dp.GPIOA.split();
        let channels = (gpioa.pa7.into_alternate(), gpioa.pa8.into_alternate());

        let pwm = dp.TIM1.pwm_hz(channels, 20.kHz(), &clocks).split();`

`error[E0277]: the trait bound `(stm32f4xx_hal::gpio::Pin<'A', 7, stm32f4xx_hal::gpio::Alternate<_>>, stm32f4xx_hal::gpio::Pin<'A', 8, stm32f4xx_hal::gpio::Alternate<_>>): stm32f4xx_hal::timer::Pins<TIM1, _>` is not satisfied
   --> src/main.rs:26:34
    |
26  |         let pwm = dp.TIM1.pwm_hz(channels, 20.kHz(), &clocks).split();
    |                           ------ ^^^^^^^^ the trait `stm32f4xx_hal::timer::Pins<TIM1, _>` is not implemented for `(stm32f4xx_hal::gpio::Pin<'A', 7, stm32f4xx_hal::gpio::Alternate<_>>, stm32f4xx_hal::gpio::Pin<'A', 8, stm32f4xx_hal::gpio::Alternate<_>>)`
    |                           |
    |                           required by a bound introduced by this call
    |
    = help: the following other types implement trait `stm32f4xx_hal::timer::Pins<TIM, P>`:
              <(P1, P2) as stm32f4xx_hal::timer::Pins<TIM, (Ch<0, NC1>, Ch<1, NC2>)>>
              <(P1, P2, P3) as stm32f4xx_hal::timer::Pins<TIM, (Ch<0, NC1>, Ch<1, NC2>, Ch<2, NC3>)>>
              <(P1, P2, P3, P4) as stm32f4xx_hal::timer::Pins<TIM, (Ch<0, NC1>, Ch<1, NC2>, Ch<2, NC3>, Ch<3, NC4>)>>
              <(P1, P2, P4) as stm32f4xx_hal::timer::Pins<TIM, (Ch<0, NC1>, Ch<1, NC2>, Ch<3, NC4>)>>
              <(P1, P3) as stm32f4xx_hal::timer::Pins<TIM, (Ch<0, NC1>, Ch<2, NC3>)>>
              <(P1, P3, P4) as stm32f4xx_hal::timer::Pins<TIM, (Ch<0, NC1>, Ch<2, NC3>, Ch<3, NC4>)>>
              <(P1, P4) as stm32f4xx_hal::timer::Pins<TIM, (Ch<0, NC1>, Ch<3, NC4>)>>
              <(P2, P3) as stm32f4xx_hal::timer::Pins<TIM, (Ch<1, NC2>, Ch<2, NC3>)>>
            and 3 others
note: required by a bound in `pwm_hz`
   --> /home/jschiefer/.cargo/git/checkouts/stm32f4xx-hal-fe8350cc04cacf3f/3593700/src/timer/pwm.rs:136:15
    |
136 |         PINS: Pins<Self, P>;
    |               ^^^^^^^^^^^^^ required by this bound in `pwm_hz`

error[E0599]: the method `split` exists for struct `PwmHz<TIM1, _, (stm32f4xx_hal::gpio::Pin<'A', 7, stm32f4xx_hal::gpio::Alternate<_>>, stm32f4xx_hal::gpio::Pin<'A', 8, stm32f4xx_hal::gpio::Alternate<_>>)>`, but its trait bounds were not satisfied
  --> src/main.rs:26:63
   |
26 |         let pwm = dp.TIM1.pwm_hz(channels, 20.kHz(), &clocks).split();
   |                                                               ^^^^^ method cannot be called on `PwmHz<TIM1, _, (stm32f4xx_hal::gpio::Pin<'A', 7, stm32f4xx_hal::gpio::Alternate<_>>, stm32f4xx_hal::gpio::Pin<'A', 8, stm32f4xx_hal::gpio::Alternate<_>>)>` due to unsatisfied trait bounds
   |
   = note: the following trait bounds were not satisfied:
           `(stm32f4xx_hal::gpio::Pin<'A', 7, stm32f4xx_hal::gpio::Alternate<_>>, stm32f4xx_hal::gpio::Pin<'A', 8, stm32f4xx_hal::gpio::Alternate<_>>): stm32f4xx_hal::timer::Pins<TIM1, _>`

error[E0283]: type annotations needed
   --> src/main.rs:24:35
    |
24  |         let channels = (gpioa.pa7.into_alternate(), gpioa.pa8.into_alternate());
    |                                   ^^^^^^^^^^^^^^ cannot infer the value of the const parameter `A` declared on the associated function `into_alternate`
    |
    = note: multiple `impl`s satisfying `stm32f4xx_hal::gpio::Pin<'A', 7>: gpio::marker::IntoAf<_>` found in the `stm32f4xx_hal` crate:
            - impl<MODE> gpio::marker::IntoAf<1> for stm32f4xx_hal::gpio::Pin<'A', 7, MODE>;
            - impl<MODE> gpio::marker::IntoAf<2> for stm32f4xx_hal::gpio::Pin<'A', 7, MODE>;
            - impl<MODE> gpio::marker::IntoAf<5> for stm32f4xx_hal::gpio::Pin<'A', 7, MODE>;
note: required by a bound in `gpio::convert::<impl stm32f4xx_hal::gpio::Pin<P, N, MODE>>::into_alternate`
   --> /home/jschiefer/.cargo/git/checkouts/stm32f4xx-hal-fe8350cc04cacf3f/3593700/src/gpio/convert.rs:121:15
    |
121 |         Self: marker::IntoAf<A>,
    |               ^^^^^^^^^^^^^^^^^ required by this bound in `gpio::convert::<impl stm32f4xx_hal::gpio::Pin<P, N, MODE>>::into_alternate`
help: consider specifying the type argument in the function call
    |
24  |         let channels = (gpioa.pa7.into_alternate::<A>(), gpioa.pa8.into_alternate());
    |                                                 +++++

`

burrbull, 2023-02-14:
try `let channels = (gpioa.pa8.into_alternate(), gpioa.pa7.into_alternate());`

Complementary pins should be passed after main.

janschiefer, 2023-02-14:
Well, I got a first success with the following code:

```
#![deny(unsafe_code)]
#![allow(clippy::empty_loop)]
#![no_main]
#![no_std]

// Halt on panic
use panic_halt as _; // panic handler

use cortex_m_rt::entry;
use stm32f4xx_hal as hal;

use crate::hal::{pac, prelude::*, timer::Channel};

#[entry]
fn main() -> ! {
    
    
      if let Some(dp) = pac::Peripherals::take() {
        // Set up the system clock.
        let rcc = dp.RCC.constrain();
        let clocks = rcc.cfgr.freeze();

        let gpioa = dp.GPIOA.split();
	let channels = (gpioa.pa8.into_alternate(), gpioa.pa7.into_alternate());

        let mut pwm = dp.TIM1.pwm_hz(channels, 20.kHz(), &clocks);
        
         let max_duty = pwm.get_max_duty();
         pwm.set_duty(Channel::C1 ,max_duty / 2);
         pwm.enable( Channel::C1 );
         pwm.enable_complementary( Channel::C1 );

         
    }

    loop {
        cortex_m::asm::nop();
    }
    
}
```

[image]

However, I need to do more intensive testing in the next days. :-D

janschiefer, 2023-02-15:
Update:

I have written a little test program, which tests most of the functions. 

```
#![deny(unsafe_code)]
#![allow(clippy::empty_loop)]
#![no_main]
#![no_std]

// Halt on panic
use panic_halt as _; // panic handler

use cortex_m_rt::entry;
use stm32f4xx_hal as hal;

use micromath::F32Ext;

use crate::hal::{pac, prelude::*, timer::Channel, timer::Polarity};

#[entry]
fn main() -> ! {
    if let Some(dp) = pac::Peripherals::take() {
        // Set up the system clock. We want to run at 84MHz for this one.
        let rcc = dp.RCC.constrain();
        let clocks = rcc.cfgr.sysclk(84.MHz()).freeze();

        let gpioa = dp.GPIOA.split();

        let channels = (gpioa.pa8.into_alternate(), gpioa.pa7.into_alternate());

        let mut pwm = dp.TIM1.pwm_hz(channels, 20.kHz(), &clocks);

        let mut wait_counter = dp.TIM2.counter_us(&clocks);

        let max_duty: u16 = pwm.get_max_duty();

        let mut test_phase: u8 = 0;

        pwm.set_duty(Channel::C1, 0);

        wait_counter.start(10.millis()).unwrap();

        loop {
            match test_phase {
                0..=10 => {
                    let pwm_duty_float: f32 = (f32::from(test_phase) * 0.1 * f32::from(max_duty))
                        .round()
                        .max(0.0)
                        .min(f32::from(max_duty));

                    pwm.enable(Channel::C1);
                    pwm.enable_complementary(Channel::C1);

                    pwm.set_polarity(Channel::C1, Polarity::ActiveHigh);
                    pwm.set_complementary_polarity(Channel::C1, Polarity::ActiveHigh);

                    pwm.set_duty(Channel::C1, pwm_duty_float as u16);
                }

                11 => {
                    pwm.set_duty(Channel::C1, max_duty / 2);

                    pwm.set_complementary_polarity(Channel::C1, Polarity::ActiveLow);
                }

                12 => {
                    pwm.set_duty(Channel::C1, max_duty / 2);

                    pwm.set_complementary_polarity(Channel::C1, Polarity::ActiveHigh);
                }

                13 => {
                    pwm.set_duty(Channel::C1, max_duty / 2);

                    pwm.set_polarity(Channel::C1, Polarity::ActiveLow);
                }

                14 => {
                    pwm.set_duty(Channel::C1, max_duty / 2);

                    pwm.set_polarity(Channel::C1, Polarity::ActiveHigh);
                }

                15 => {
                    pwm.set_duty(Channel::C1, max_duty / 2);

                    pwm.set_polarity(Channel::C1, Polarity::ActiveHigh);
                    pwm.set_complementary_polarity(Channel::C1, Polarity::ActiveHigh);

                    pwm.enable(Channel::C1);
                    pwm.disable_complementary(Channel::C1);
                }

                16 => {
                    pwm.set_duty(Channel::C1, max_duty / 2);

                    pwm.set_polarity(Channel::C1, Polarity::ActiveHigh);
                    pwm.set_complementary_polarity(Channel::C1, Polarity::ActiveHigh);

                    pwm.disable(Channel::C1);
                    pwm.enable_complementary(Channel::C1);
                }

                _ => {}
            }

            if test_phase < 16 {
                test_phase += 1;
            } else {
                test_phase = 0;
            }

            nb::block!(wait_counter.wait()).unwrap();
        }
    }

    loop {
        cortex_m::asm::nop();
    }
}
```

All the tested cases seem to work just fine ( tested on STM401 "BlackPill" development board).

Attachments:
- Full test program repository: [stm32_401_complementary_pwm.zip](https://github.com/stm32-rs/stm32f4xx-hal/files/10747160/stm32_401_complementary_pwm.zip)

- Logic analyzer capture ( PulseView ): 
[stmf401_pwm_logic_analyzer_capture.zip](https://github.com/stm32-rs/stm32f4xx-hal/files/10747193/stmf401_pwm_logic_analyzer_capture.zip)

Very nice work so far!!! :+1: 

However, what is still missing for a fully useful implementation:
- Set CH and CHN Idle State ( ActiveLow or AcriveHigh )
- Dead time generation ( This is vital for real world scenarios. )
- A good example ( Maybe I can write up a nice example... )
- Good Documentation 
( Please explain why" let channels = (gpioa.pa8.into_alternate(), gpioa.pa7.into_alternate()) " works and "let channels = (gpioa.pa7.into_alternate(), gpioa.pa8.into_alternate()) " does not... )

burrbull, 2023-02-16:
> A good example ( Maybe I can write up a nice example... )

Would be nice.

> Dead time generation ( This is vital for real world scenarios. )

TIM1 specific, yes? What API do you suggest?

> Set CH and CHN Idle State ( ActiveLow or AcriveHigh )

Not you I understand what you mean. MOE bit?

burrbull, 2023-02-16:
what about this https://github.com/stm32-rs/stm32f4xx-hal/commit/fff8d6026025b3f05bf57ed7a200c2c13f65a1c0 ?

janschiefer, 2023-02-16:
Sorry. I formulated it wrong.

I mean he polarity for CH and CHN Idle State which should be "IdleHigh" or "IdleLow".

[image]

I will test the commit https://github.com/stm32-rs/stm32f4xx-hal/commit/9a2ff702095bab582c3615ba93b62b9d5fefceaa  in the next days for dead time generation.

burrbull, 2023-02-16:
Oh. I understand. OIS bits. 
First state is `Reset`. What name of second in Cube?

janschiefer, 2023-02-16:
Reset / Set.

burrbull, 2023-02-18:
What rounding to use for dead time, ceil or floor?
https://play.rust-lang.org/?version=stable&mode=debug&edition=2021&gist=41148a1b85c4ad656635d913b593c6e1

janschiefer, 2023-02-18:
The compiler complains about missing &self attribute for the new function set_dead_time(). from the commit https://github.com/stm32-rs/stm32f4xx-hal/commit/9a2ff702095bab582c3615ba93b62b9d5fefceaa .

How do I use this?

```

let rcc = dp.RCC.constrain();
let clocks = rcc.cfgr.sysclk(84.MHz()).freeze();

let gpioa = dp.GPIOA.split();

let channels = (gpioa.pa8.into_alternate(), gpioa.pa7.into_alternate());

let mut pwm = dp.TIM1.pwm_hz(channels, 20.kHz(), &clocks);

let max_duty: u16 = pwm.get_max_duty();

pwm.set_duty(Channel::C1, max_duty / 2);

pwm.set_polarity(Channel::C1, Polarity::ActiveHigh);
pwm.set_complementary_polarity(Channel::C1, Polarity::ActiveHigh);

pwm.set_dead_time(50);
pwm.enable(Channel::C1);
pwm.enable_complementary(Channel::C1);
```

```
error[E0599]: no method named `set_dead_time` found for struct `PwmHz<TIM1, Ch<0, true>, (stm32f4xx_hal::gpio::Pin<'A', 8, stm32f4xx_hal::gpio::Alternate<1>>, stm32f4xx_hal::gpio::Pin<'A', 7, stm32f4xx_hal::gpio::Alternate<1>>)>` in the current scope
   --> src/main.rs:106:11
    |
106 |             pwm.set_dead_time(50);
    |             ----^^^^^^^^^^^^^----
    |             |   |
    |             |   this is an associated function, not a method
    |             help: use associated function syntax instead: `PwmHz::<TIM1, Ch<0, true>, (stm32f4xx_hal::gpio::Pin<'A', 8, stm32f4xx_hal::gpio::Alternate<1>>, stm32f4xx_hal::gpio::Pin<'A', 7, stm32f4xx_hal::gpio::Alternate<1>>)>::set_dead_time(pwm, 50)`
    |
    = note: found the following associated functions; to be used as methods, functions must have a `self` parameter
    = note: the candidate is defined in an impl for the type `PwmHz<TIM, P, PINS>`

For more information about this error, try `rustc --explain E0599`.

```

Can I somehow help you with determining the rounding mode? ( Maybe testing some C code in STM32CubeIDE... )

burrbull, 2023-02-19:
Those commit is broken. See #578 instead

janschiefer, 2023-02-27:
Commit https://github.com/stm32-rs/stm32f4xx-hal/pull/578 looks fine so far.
```
       
        let rcc = dp.RCC.constrain();
        let clocks = rcc.cfgr.sysclk(84.MHz()).freeze();

        let gpioa = dp.GPIOA.split();

        let channels = (gpioa.pa8.into_alternate(), gpioa.pa7.into_alternate());

        let mut pwm = dp.TIM1.pwm_hz(channels, 20.kHz(), &clocks);

        let max_duty: u16 = pwm.get_max_duty();

        pwm.set_duty(Channel::C1, max_duty / 2);

        pwm.set_polarity(Channel::C1, Polarity::ActiveHigh);
        pwm.set_complementary_polarity(Channel::C1, Polarity::ActiveHigh);

        pwm.set_dead_time(100);
        pwm.enable(Channel::C1);
        pwm.enable_complementary(Channel::C1);

```

[image]

burrbull, 2023-02-27:
Is it what you expected?

janschiefer, 2023-02-27:
I also played around with the DTG bitfield, and it seemd to work as expected.

```
pwm.set_dead_time( 0b00011111 );
(...)
pwm.set_dead_time( 0b11111111 );
```

[image]

burrbull, 2023-02-27:
> pwm.set_dead_time( 0b00011111 );

The idea was not to calculate bits manually.
You just pass time in DTS ticks (any value from 0 to 1008) and it rounds it to nearest ceiling value that can be converted to bits.

janschiefer, 2023-03-01:
This is actually nice way of simplifying the original dead time configuration from the official C/C++ STM 32 SDK a bit.

However, since this differs significantly from original approach I think this has to be explained in the documentation.

TIM::set_dtg_value(bits) should be documented as the alternative way if someone prefers a more "bare metal" way of configuration.

janschiefer, 2023-03-01:
Maybe even a function that takes a "TimerDurationU32<FREQ>" argument and rounds to the nearest possible value can be implemented in the future to bypass the original configuration workflow.

burrbull, 2023-03-01:
I thought about this, but I don't think this can be implemented easy.

For `fugit::Duration` you need to specify FREQ in compile time as those are different types.
Even if you think about some fixed granularity, for example nanoseconds, you still need to know timer clock frequency and also read `CR1.CKD`. And this conversion will not zero cost.

janschiefer, 2023-03-03:
Sure. I'll write something.

janschiefer, 2023-03-13:
Complete creates.io documentation pull request: https://github.com/stm32-rs/stm32f4xx-hal/pull/584
Dead time insertion example pull request: https://github.com/stm32-rs/stm32f4xx-hal/pull/585