stm32-rs/stm32f4xx-hal #550: how to access `serial::Rx::clear_idle_interrupt` when using DMA?

rursprung, 2022-11-07:
i'm using USART1 with DMA and have enabled the line idle interrupt (using [serial::Rx::listen_idle](https://docs.rs/stm32f4xx-hal/latest/stm32f4xx_hal/serial/struct.Rx.html#method.listen_idle)).
this means that i now get a `USART1` interrupt every time a message is received (the DMA buffer is generally not full at that point but i want to act on the message as soon as it has arrived completely). when acting on the interrupt i need to clear the interrupt flag. _theoretically_, there's a nice API for this: [serial::Rx::clear_idle_interrupt](https://docs.rs/stm32f4xx-hal/latest/stm32f4xx_hal/serial/struct.Rx.html#method.clear_idle_interrupt). _however_: as i have passed my `Rx` into `dma::Transfer` and that does not offer an API to get back to the `Rx` i have no way of calling it.
my quick-win for now was to just copy the implementation (also meaning i now have `unsafe` code in my crate 😞).

is there a proper way to clear the idle interrupt in this situation using existing APIs? or is an API missing for that?

for reference: this is on a STM32F401RE, but the question applies to the HAL crate in general (none of the code is device-specific as far as i could see)

ololoshka2871, 2022-11-08:
Same problem with SPI.
Then i build `stm32f4xx_hal::dma::Transfer` it consumes `stm32f4xx_hal::spi::DmaBuilder` what consumes SPI object `stm32f4xx_hal::spi::Spi::use_dma(self)`.
No idea how to access to SPI object.

rursprung, 2022-12-07:
> No idea how to access to SPI object.

i just checked and didn't see a corresponding `clear_idle_interrupt` (or any other interrupt clearing) API for `Spi`. what functionality are you looking for?

rursprung, 2022-12-09:
this has been solved with #556