stm32-rs/stm32f4xx-hal #567: Fail to set up AHB clock (for USB) in nightly

hacknus, 2023-01-09:
I tried both USB-examples on my STM32F405; [poll](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/examples/usb-serial-poll.rs) and [interrupt](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/examples/usb-serial-irq.rs
) and both work with rust stable (rustc 1.66.0 ) but hang with nightly (rustc 1.68.0-nightly).  
By using the debugger, I notice that I get stuck in this line:
```
// Wait for AHB ready
while read_reg!(otg_global, regs.global(), GRSTCTL, AHBIDL) == 0 {}
```
line 365 in `.cargo/registry/src/github.com-1ecc6299db9ec823/synopsys-usb-otg-0.3.1/src/bus.rs`

full output of the debugger:

```
synopsys_usb_otg::bus::{impl#2}::enable::{closure#0}<stm32f4xx_hal::otg_fs::USB> (cs=<optimized out>)
    at /Users/linus/.cargo/registry/src/github.com-1ecc6299db9ec823/synopsys-usb-otg-0.3.1/src/bus.rs:365
365	            while read_reg!(otg_global, regs.global(), GRSTCTL, AHBIDL) == 0 {}
(gdb) 
halted: PC: 0x08002ace
halted: PC: 0x08002ac6
synopsys_usb_otg::ral::register::RWRegister<u32>::read<u32> (self=<optimized out>)
    at /Users/linus/.cargo/registry/src/github.com-1ecc6299db9ec823/synopsys-usb-otg-0.3.1/src/ral/register.rs:23
23	        unsafe { ::core::ptr::read_volatile(self.register.get()) }
(gdb) step
0x08002ac6 in core::ptr::read_volatile<u32> (src=<optimized out>) at /rustc/659e169d37990b9c730a59a96081f2ef7afbe8f1/library/core/src/ptr/mod.rs:1521
1521	/rustc/659e169d37990b9c730a59a96081f2ef7afbe8f1/library/core/src/ptr/mod.rs: No such file or directory.
(gdb) step
halted: PC: 0x08002aca
synopsys_usb_otg::bus::{impl#2}::enable::{closure#0}<stm32f4xx_hal::otg_fs::USB> (cs=<optimized out>)
    at /Users/linus/.cargo/registry/src/github.com-1ecc6299db9ec823/synopsys-usb-otg-0.3.1/src/bus.rs:365
365	            while read_reg!(otg_global, regs.global(), GRSTCTL, AHBIDL) == 0 {}
(gdb) 
halted: PC: 0x08002ace
halted: PC: 0x08002ac6
```

burrbull, 2023-01-10:
https://github.com/stm32-rs/synopsys-usb-otg

adamgreig, 2023-01-14:
Closing this issue in preference of https://github.com/stm32-rs/synopsys-usb-otg/issues/33, thanks for reporting!

closed 2023-01-14 by adamgreig