stm32-rs/stm32f4xx-hal #570: Timer PWM cannot compile when initializing multiple pins/channels on different ports.

hacknus, 2023-01-16:
I am working on an STM32F405. The following code works:
```
    let mut tim3_pwm = timer::Timer::new(dp.TIM3, &clocks).pwm_hz(
                                          gpiob.pb1.into_alternate()
                                      , 2000.Hz());
```
so does this:
```
    let mut tim3_pwm = timer::Timer::new(dp.TIM3, &clocks).pwm_hz(
                                          gpioc.pc6.into_alternate()
                                      , 2000.Hz());
```
and this:
```
    let mut tim3_pwm = timer::Timer::new(dp.TIM3, &clocks).pwm_hz(
                                          gpioc.pc7.into_alternate()
                                      , 2000.Hz());
```
and this:
```
    let mut tim3_pwm = timer::Timer::new(dp.TIM3, &clocks).pwm_hz((
                                          gpioc.pc6.into_alternate(),
                                          gpioc.pc7.into_alternate()
                                      ), 2000.Hz());
```

however, when using multiple ports (GPIOB and GPIOC) it cannot compile:
```
    let mut tim3_pwm = timer::Timer::new(dp.TIM3, &clocks).pwm_hz((
                                          gpiob.pb1.into_alternate(),
                                          gpioc.pc6.into_alternate(),
                                          gpioc.pc7.into_alternate()
                                      ), 2000.Hz());
```
with the error:
```
60  |       let mut tim3_pwm = timer::Timer::new(dp.TIM3, &clocks).pwm_hz((
    |  ____________________________________________________________------_^
    | |                                                            |
    | |                                                            required by a bound introduced by this call
61  | |                                           gpiob.pb1.into_alternate(),
62  | |                                           gpioc.pc6.into_alternate(),
63  | |                                           gpioc.pc7.into_alternate()
64  | |                                       ), 2000.Hz());
    | |_______________________________________^ the trait `stm32f4xx_hal::timer::Pins<stm32f4xx_hal::pac::TIM3, _>` is not implemented for `(stm32f4xx_hal::gpio::Pin<'B', 1, stm32f4xx_hal::gpio::Alternate<_>>, stm32f4xx_hal::gpio::Pin<'C', 6, stm32f4xx_hal::gpio::Alternate<_>>, stm32f4xx_hal::gpio::Pin<'C', 7, stm32f4xx_hal::gpio::Alternate<_>>)`
    |
    = help: the following other types implement trait `stm32f4xx_hal::timer::Pins<TIM, P>`:
              <(P1, P2) as stm32f4xx_hal::timer::Pins<TIM, (Ch<0>, Ch<1>)>>
              <(P1, P2, P3) as stm32f4xx_hal::timer::Pins<TIM, (Ch<0>, Ch<1>, Ch<2>)>>
              <(P1, P2, P3, P4) as stm32f4xx_hal::timer::Pins<TIM, (Ch<0>, Ch<1>, Ch<2>, Ch<3>)>>
              <(P1, P2, P4) as stm32f4xx_hal::timer::Pins<TIM, (Ch<0>, Ch<1>, Ch<3>)>>
              <(P1, P3) as stm32f4xx_hal::timer::Pins<TIM, (Ch<0>, Ch<2>)>>
              <(P1, P3, P4) as stm32f4xx_hal::timer::Pins<TIM, (Ch<0>, Ch<2>, Ch<3>)>>
              <(P1, P4) as stm32f4xx_hal::timer::Pins<TIM, (Ch<0>, Ch<3>)>>
              <(P2, P3) as stm32f4xx_hal::timer::Pins<TIM, (Ch<1>, Ch<2>)>>
            and 3 others
note: required by a bound in `pwm::<impl Timer<TIM>>::pwm_hz`
   --> /Users/linus/.cargo/git/checkouts/stm32f4xx-hal-fe8350cc04cacf3f/15134d7/src/timer/pwm.rs:232:15
    |
232 |         PINS: Pins<TIM, P>,
    |               ^^^^^^^^^^^^ required by this bound in `pwm::<impl Timer<TIM>>::pwm_hz`
```

The same error occurs for a slightly different syntax:
```
    let mut tim3_pwm = dp.TIM3.pwm_hz((
                                          gpiob.pb1.into_alternate(),
                                          gpioc.pc6.into_alternate(),
                                          gpioc.pc7.into_alternate()
                                      ), 2000.Hz(), &clocks);
```

burrbull, 2023-01-17:
`(PC6, PC7, PB1)` order should work.

You are right. Pwm part of hal need deep rework. Especially initialization. Possibly with builder pattern.
But it requires much time I don't have.

hacknus, 2023-01-17:
Yes, that works, thank you! 
And in general, thanks so much for your contribution, I am really enjoying working with the `stm32f4xx` crate!

burrbull, 2023-01-17:
What reason you prefer `PwmHz` over `Pwm`?

hacknus, 2023-01-17:
it's what I found in `stm32f4xx-hal/examples` [pwm.rs](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/examples/pwm.rs) and [a buzzer example](https://dev.to/apollolabsbin/stm32f4-embedded-rust-at-the-hal-pwm-buzzer-3f1b).

closed 2023-03-13 by burrbull