stm32-rs/stm32f4xx-hal #574: USB fails without any errors on nightly

t348575, 2023-01-30:
USB appears to fail when the `UsbDeviceBuilder` is called, regardless of what it is used for (serial, hid etc). The device gets stuck at this line and never goes over (does not panic, just gets stuck). Not sure if this is an issue with `stm32f4xx-hal` or with [usb-device](https://github.com/rust-embedded-community/usb-device), and it is happening only on nightly, all usb forms run fine on stable. 

## Line that fails: https://github.com/stm32-rs/stm32f4xx-hal/blob/15134d7063083ec246302b09f18eb3c774fd1f5e/examples/usb-serial-poll.rs#L43

## Environment:
* stm32f401ccu6 (black pill)
* nighly version (failing): 1.69.0
* stable version (working): 1.67.0

burrbull, 2023-01-30:
It could also be related to https://github.com/stm32-rs/synopsys-usb-otg/issues/33

Try `cargo update`, please.

t348575, 2023-01-30:
@burrbull Yep that worked.

closed 2023-01-30 by t348575