stm32-rs/stm32f4xx-hal #592: Support half-duplex serial

llamington, 2023-04-02:
Currently the `serial` function requires a tuple of RX/TX pins. The [reference manual](https://www.st.com/resource/en/reference_manual/rm0390-stm32f446xx-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) states that "the TX and RX lines are internally connected", so only one pin would be required as an argument.

burrbull, 2023-04-02:
You mean both read and write on TX line?

llamington, 2023-04-03:
Yes that's correct

burrbull, 2023-04-03:
Not implemented yet.

It's better to create independent `SerialHalfDuplex` structure for this with `new::(pin: TX)` and its own methods.
As I understand it will be semantically incompatible with `read`/`write` from `embedded-hal`.

bluespider42, 2024-06-05:
@llamington Did you ever make any progress on this?

llamington, 2024-06-05:
@bluespider42 no, unfortunately I didn't (I ended up using the STM32HAL library in C)

bluespider42, 2024-06-19:
embassy-stm32 hal has a half-duplex uart.

bluespider42, 2024-10-21:
https://github.com/stm32-rs/stm32f4xx-hal/pull/781