stm32-rs/stm32f4xx-hal #619: What's BSP?

TheFern2, 2023-05-02:
Sorry if this is a silly question, new to rust development 

> Check if the BSP for your board exists in the stm32-rs page. If it exists, the stm32f4xx-hal crate should be already included, so you can use the bsp as BSP for your project.

What's the meaning of this paragraph?

I have a sample setup for my nucleo https://github.com/TheFern2/test_nucleo when using cargo embed I get a warning about using custom setup for the board stm324266retx. Code works though just wondering if I'm doing something wrong.

burrbull, 2023-05-02:
`BSP` is board support package. Usually small crates with additional wrappers for board. For example:
https://github.com/stm32-rs/stm32f407g-disc/

burrbull, 2023-05-02:
> `WARN probe_rs::config::target > Using custom sequence for ARMv7 STM32F446RETx`

This is unrelated to those phrase.
It just says that `probe-rs` can change some registers state during flashing. For example enable DBG clocking in sleep mode.