stm32-rs/stm32f4xx-hal #643: [Possible user error] Clock speed too low stm32f401

dvdsk, 2023-05-19:
### Context
For one of my hobby projects own usb project (reading/spying? usb as a third party). It uses an stm32f401 (black pill board). But now I am starting to doubt myself. I have set the mcu to run at 84 Mhz, Usb full speed runs at 12 so that should give me 7 clock cycles for each usb bit. I wrote an interrupt routine in armv7 assembly that aquires a bunch of samples (loop unrolled). I padded each gpio read  (load instruction) with nop's such that the period between samples is 7 clock cycles. I also made sure the interrupt is running from .data (so it is kept in RAM instead of flash). 

This didn't really seem to work. Yesterdays I got the idea to put a pin high at the start of the interrupt (and low at the end) to time the whole routine. The timing was way off, a factor 5. This gave me the idea that something might be wrong with the clock. 

### Issue
So I made an interrupt routine in assembly which pulls a gpio high then goes through 100 nops before putting the gpio low.  
 - at 84 Mhz a NOP should take one clock cycle or ~= 12 ns
-  100 NOPS should take 1.2 micro seconds.

I monitor the GPIO using a cheap 24Mhz logic analyzer. It measures a period of 6.3 micro seconds. In the screenshot below you see the results, on D6 the debug pulse of 100 nops.
[image]

I also used MCO1 to output the pll with a div of 5. That should output a lock signal of about 17 Mhz. Since the logic analyzer can only sample to 24 Mhz I would expect to read 24 or 17 Mhz. Instead it finds 3 (see screenshot below D7).
[image]

### Clock Settings and source
TBD (submitted as draft issue)

burrbull, 2023-05-20:
`rcc.cfgr.use_hse(25.MHz()).sysclk(84.MHz()).freeze()`

dvdsk, 2023-05-20:
> `rcc.cfgr.use_hse(25.MHz()).sysclk(84.MHz()).freeze()`

Unfortunately that (and `use_hse(16.MHz())`) does not work. I have updated the issue/question to reflect me trying that.

dvdsk, 2023-05-21:
Update: today I got MCO pll out working in C via stm32cube. With the firmware setup to run at 84Mhz. The PLL is now significantly higher

dvdsk, 2023-05-21:
At this point I could use some guidance on how to find what specifically is going wrong on the Rust side. Is there anyone who could mentor me in figuring this out? I suspect something might be going wrong in stm32fxx-hal when configuring the clock, no idea how to confirm that though.

dvdsk, 2023-05-23:
Update: today I found out I made a mistake configuring MCO1. The clockspeed is set correctly by stm32f4xx-hal. I will leave this issue open while I continue to look into why the interrupt of NOPS takes longer then expected.

dvdsk, 2023-05-24:
>         (*hal::pac::RCC::ptr()).cfgr.write(|w| {
>             w.mco1().pll();
>             w.mco1pre().div5()
>         });

Configuring the MCO like this overwrites some clock configuration. As `write` resets fields to their default value. It looks like me misconfiguring the MCO was the cause of all my troubles here. Ill double check with my original code base and then close the issue.

closed 2023-05-24 by dvdsk

dvdsk, 2024-06-05:
> @dvdsk Did you get it working correctly using
> 
> ```rust
> cfgr.modify|_,w| {
>     w.mco1().pll();
>     w.mco1pre().div5()
> });
> ```
> 
> instead? or something similar?

yes, that gave me correct mco output