stm32-rs/stm32f4xx-hal #644: does dma::StreamX miss some method ?

YruamaLairba, 2023-05-26:
Hi,
To me, StreamX seems to be just a thin wrapper to control DMA stream without trying to give a safe abstraction. However some  setup are missing, i wonder if there is a reason:
- PINCOS: peripheral increment offset size
- CIRC: circular mode (i'm particularly interested by this one :) )
- PFCTRL: peripheral flow controller

from API choice point of view i have remarks about "set_channel" and "set_direction"  they both use generic while an enum seems enough to me. I understand it's to work with typestate marker system, and you can probably play with the typestate marker system without having a generic here. Maybe it's because StreamX wasn't initially designed to be used outside the crate ?

burrbull, 2023-05-26:
Base of this API was made quite a long time ago. A I don't know it well. I would also want to see circular API. Not sure how to add it.
Maybe H7 trait can be used https://github.com/stm32-rs/stm32h7xx-hal/blob/master/src/dma/traits.rs. Although DMA structure is quite different.

I don't see why you need enums in `set_channel`. Except for a few exceptions (TIM1_CCR3) there is only 1 channel that can be used with specific combination of peripheral/stream. See tables in RM "DMA functional description". So this prevents you passing incorrect channel.

YruamaLairba, 2023-05-26:
> So this prevents you passing incorrect channel.

This is not true when you manipulate the StreamX directly

> I don't see why you need enums in set_channel

This is not what i say, I just wonder why to use generic here when an enum could be enought. I understood this is probably to prevent bad configuration elsewhere, but i think you could also prevent misconfiguration without having generic here.

For PINCOS, CIRC and PFCTRL, i don't see what is the issue to have it on StreamX since this type require unsafe to be usable.

burrbull, 2023-05-27:
Do I understand right that you want to use DMA configuration in independent crate? It looks impossible.
DMA stuff is very different across families. Only F2 and F7 are similar to F4.
[image]

Only transfer traits can be universal enough to interact between crates. https://docs.rs/embedded-dma/ 
Configuration is too specific to move it out of HAL.

YruamaLairba, 2023-05-27:
Where did you get the infos about version of DMA used across stm32 families ?

> Do I understand right that you want to use DMA configuration in independent crate?

For the moment, I want to use DMA configuration in a project, not in a general purpose HAL crate. Ideally, I want to use `StreamX` with circular mode configuration implemented because it's nicer to use than PAC stuff. `Transfer` is not suitable. for what i want to do. I want to read/write a buffer while 2 DMA Stream access to it (one Stream reading, the other Writing). I think this kind of operation can have a "Safe" interface if you use (or implement) appropriate Atomic type for the buffer. But for know i just want to make it work even if it's unsound :p 

> Configuration is too specific to move it out of HAL

I agree :smile:

burrbull, 2023-05-27:
> Where did you get the infos about version of DMA used across stm32 families ?

https://docs.google.com/spreadsheets/d/1-R-AjYrMLL2_623G-AFN2A9THMf8FFMpFD4Kq-owPmI

YruamaLairba, 2023-05-27:
Weird, for SPI, this document doesn't follow ST numbering for version. for ST, it's v1.2, v1.3, v2 and V3

YruamaLairba, 2023-05-30:
i did a draft pull request #650 to show you that generic method of Stream could be replaced by  non-generic while keep Working the Transfer type.
I may did something wrong wheen rebasing, the history is weird

everdrone, 2023-10-08:
I was also looking for the circular mode option. Any update on this?

YruamaLairba, 2023-10-08:
@everdrone i rewrote the Stream API sometimes ago. This include the circular mode option, and since my PR was merged ( #666) this will probably available in the next release.

burrbull, 2024-06-01:
Closing. Open new one if not completed.

closed 2024-06-01 by burrbull