stm32-rs/stm32f4xx-hal #646: use `gpio::ErasedPin::restore` without generic parms?

eZioPan, 2023-05-28:
Let's says there is a 8 pins, I want to write to them first, then read from them (eg. LCD1602 use same 8 pins to read instructions and output busy state / memory content).

As `gpio::Pin::erase` says, it could be easy to store a list of `ErasedPin` into a list. But when I need to change pin `Mode` (from Output to Input), I will need to `restore` pins. But `restore`  function ask about Port name and pin Number, which are missed.

If it's possible that `restore` could auto-infer the pin information, so that `erase` and list of `ErasedPin` can have broader usage?

sorry for my poor example code:

```rust
#![no_main]
#![no_std]

use panic_halt as _;

use stm32f4xx_hal::{
    gpio::{ErasedPin, Output, PinState},
    pac,
    prelude::*,
};

#[cortex_m_rt::entry]
fn main() -> ! {
    let dp = pac::Peripherals::take().expect("Cannot take device peripherals");

    let rcc = dp.RCC.constrain();

    let _clock = rcc.cfgr.freeze();

    let gpioa = dp.GPIOA.split();
    let gpiob = dp.GPIOB.split();

    let pin0 = gpioa
        .pa0
        .into_push_pull_output_in_state(PinState::Low)
        .erase();
    let pin1 = gpiob
        .pb1
        .into_push_pull_output_in_state(PinState::Low)
        .erase();

    //  we need to move pins out of an array when modify pin Mode,
    // `Option<ErasedPin>` could be sufficient to satisfy borrow check.
    let mut pins = [Some(pin0), Some(pin1)];

    write_to_pins(&mut pins);

    read_from_pins(&mut pins);

    loop {}
}

// `Option<ErasedPin>` for modify pin Mode, it's not useful if we don't need to move pins out of an array
fn write_to_pins(pins: &mut [Option<ErasedPin<Output>>]) {
    for pin in pins.iter_mut() {
        let mut temp_pin = pin.take().unwrap();
        temp_pin.set_high();
        pin.replace(temp_pin);
    }
}

fn read_from_pins(pins: &mut [Option<ErasedPin<Output>>]) -> [bool; 2] {
    let mut result: [bool; 2] = [false; 2];

    for (index, pin) in pins.iter_mut().enumerate() {
        let mut temp_pin = pin.take().unwrap();
        // TODO:
        // How to make this possible?
        // let temp_pin = temp_pin.restore<?,?>().into_input();
        // bits[index] = temp_pin.is_high();
        // let temp_pin = temp_pin.into_pull_push_output().erase();
        pin.replace(temp_pin);
    }
    result
}
```

burrbull, 2023-05-28:
Possibly we could implement changing mode for `ErasedPin`s.
But it is impossible to infer type for `restore`. `ErasedPin` "erases" type.

It is more convenient to use `OutputPort` or `[dyn OutputPin]` to write several lines at time.
I hear first time you can read something from LCD1602. (Do you need `Input<Floating>` mode or maybe `Output<OpenDrain`?)

eZioPan, 2023-05-29:
> Possibly we could implement changing mode for `ErasedPin`s.

This will be good enough, i believe it will cover most use case.

> It is more convenient to use `OutputPort` or `[dyn OutputPin]` to write several lines at time.

Ah, thanks for pointing out, i will take a look at the document.

> I hear first time you can read something from LCD1602.

There are 2 instructions called **Read busy flag & address** and **Read data from CG or DDRAM** in both `HD44780U` and `ST7066U`, that we can pull some data from LCD1602.

> (Do you need `Input<Floating/PullX>` mode or maybe `Output<OpenDrain>`?)

I totally forget about `Output<OpenDrain>`. I will use it for now, thanks.

burrbull, 2023-05-29:
> This will be good enough, i believe it will cover most use case.

https://github.com/stm32-rs/stm32f4xx-hal/pull/647

closed 2023-06-10 by eZioPan