stm32-rs/stm32f4xx-hal #651: language item required, but not found: `eh_personality`

GenesisAN, 2023-06-01:
I encountered this error after running the cargo build command while following the steps in the README to create a project for STM32F407. I'm not sure why this error is happening. Here are some of my code files, most of which are from the project's README, and I haven't made many modifications.

``` ps
PS D:\Code\Rust\stm32f407app> cargo build  
   Compiling stm32f407app v0.1.0 (D:\Code\Rust\stm32f407app)
error: language item required, but not found: `eh_personality`
  |
  = note: this can occur when a binary crate with `#![no_std]` is compiled for a target where `eh_personality` is defined in the standard library
  = help: you may be able to compile for a target that doesn't need `eh_personality`, specify a target with `--target` or in `.cargo/config`

error: could not compile `stm32f407app` due to previous error
```

main.rs:
``` rust
#![deny(unsafe_code)]
#![allow(clippy::empty_loop)]
#![no_main]
#![no_std]

use panic_halt as _;
use cortex_m_rt::entry;
use stm32f4xx_hal as hal;

use crate::hal::{pac, prelude::*};

#[entry]
fn main() -> ! {
    if let (Some(dp), Some(cp)) = (
        pac::Peripherals::take(),
        cortex_m::peripheral::Peripherals::take(),
    ) {
        let gpioa = dp.GPIOA.split();
        let mut led = gpioa.pa5.into_push_pull_output();
        let rcc = dp.RCC.constrain();
        let clocks = rcc.cfgr.sysclk(48.MHz()).freeze();
        let mut delay = cp.SYST.delay(&clocks);
        loop {
            led.toggle();
            delay.delay_ms(1000_u32);
        }
    }

    loop {}
}

```
cargo.toml:
``` toml
[package]
name = "stm32f407app"
version = "0.1.0"
edition = "2021"

[dependencies]
embedded-hal = "0.2"
nb = "1"
cortex-m = "0.7"
cortex-m-rt = "0.7"
panic-halt = "0.2.0"

[dependencies.stm32f4xx-hal]
version = "0.16.0"
features = ["stm32f407"] 
```
.config/config:
``` toml
[target.thumbv7em-none-eabihf]
# runner = 'probe-run --chip STM32F411CEUx'
rustflags = [
  # `flip-link` moves stack at the end of flash
  #"-C", "linker=flip-link",
  "-C", "link-arg=-Tlink.x",
  "-C", "link-arg=-Tdefmt.x",
  
  # This is needed if your flash or ram addresses are not aligned to 0x10000 in memory.x
  # See https://github.com/rust-embedded/cortex-m-quickstart/pull/95
  #"-C", "link-arg=--nmagic",
]

[build]
target = "thumbv7em-none-eabihf"

[env]
DEFMT_LOG = "info"
```
memory.x
```
MEMORY
{
  /* NOTE K = KiBi = 1024 bytes */
  FLASH : ORIGIN = 0x08000000, LENGTH = 128K 
  RAM : ORIGIN = 0x20000000, LENGTH = 32K
}

/* This is where the call stack will be allocated. */
/* The stack is of the full descending type. */
/* NOTE Do NOT modify `_stack_start` unless you know what you are doing */
_stack_start = ORIGIN(RAM) + LENGTH(RAM);
```

burrbull, 2023-06-01:
Add this at the end of `Cargo.toml` and compile in release: `cargo build --release`
```
# Set the default for dependencies.
[profile.dev.package."*"]
opt-level = "s"

[profile.release]
codegen-units = 1
incremental = false
debug = true
lto = true
opt-level = "s"
```

Comment `"#-C", "link-arg=-Tdefmt.x",` if you don't use `defmt`.

GenesisAN, 2023-06-01:
Thank you for your response. It seems like it didn't work, and the error still persists. Do you need me to provide more information?

``` cmd
PS D:\Code\Rust\stm32f407app> cargo build --release
   Compiling stm32f407app v0.1.0 (D:\Code\Rust\stm32f407app)
error: language item required, but not found: `eh_personality`
  |
  = note: this can occur when a binary crate with `#![no_std]` is compiled for a target where `eh_personality` is defined in the standard library
  = help: you may be able to compile for a target that doesn't need `eh_personality`, specify a target with `--target` or in `.cargo/config`

error: could not compile `stm32f407app` due to previous error
```

cargo.toml:
``` toml
[package]
name = "stm32f407app"
version = "0.1.0"
edition = "2021"

[dependencies]
embedded-hal = "0.2"
nb = "1"
cortex-m = "0.7"
cortex-m-rt = "0.7"
panic-halt = "0.2.0"

[dependencies.stm32f4xx-hal]
version = "0.16.0"
features = ["stm32f407"]

[profile.dev.package."*"]
opt-level = "s"

[profile.release]
codegen-units = 1
incremental = false
debug = true
lto = true
opt-level = "s"

```

.config/confi.toml：
``` toml
[target.thumbv7em-none-eabihf]
rustflags = [
  "-C",
  "link-arg=-Tlink.x",
]

[build]
target = "thumbv7em-none-eabihf"

[env]
DEFMT_LOG = "info"

```

GenesisAN, 2023-06-01:
> Rename `.config/confi.toml` to `.config/config.toml`

I just made a typo. The actual file name is config.toml.

closed 2023-06-01 by GenesisAN

burrbull, 2023-06-01:
Try this simple template generator:
https://github.com/burrbull/stm32-template/

```sh
cargo generate --git https://github.com/burrbull/stm32-template/
```