stm32-rs/stm32f4xx-hal #654: Example pwm.rs from v0.16.0 does not work on F401

KarolNi, 2023-06-20:
Hi everybody!
Thanks for this library. I've just started learning rust and wanted to start from the most recent versions of libraries.
I tried to run pwm.rs example and it compiles (without any warning), but there is no activity on PA8 (or PA9) (checked by oscilloscope). After couple tries, I switched to ^0.15.0 and it worked just like that.
So there is some error in example for 0.16 or some issue with the library.

burrbull, 2023-06-20:
Could you try before/after https://github.com/stm32-rs/stm32f4xx-hal/pull/604 ?

KarolNi, 2023-06-21:
Copied and pasted pwm.rs as main.rs to my project and rev="f1b44d625e0fbfbba3dc11d53901acdd64026638" does not work.
rev="af887fec64f4794feda803e6bbc3f4b7fa8034c2" with matching pwm.rs works (square wave on oscilloscope at PA8).
Cargo.toml
``` toml
[package]
name = "dc_dyno"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[profile.release]
opt-level = 'z' # turn on maximum optimizations. We only have 64kB
lto = true      # Link-time-optimizations for further size reduction

[dependencies]
cortex-m = "^0.7.7"      # Access to the generic ARM peripherals
cortex-m-rt = "^0.7.3"  # Startup code for the ARM Core
embedded-hal = "^0.2.7"  # Access to generic embedded functions (`set_high`)
panic-halt = "^0.2.0"    # Panic handler

[dependencies.stm32f4xx-hal]
features = ["stm32f401", "rt"]
git = "https://github.com/stm32-rs/stm32f4xx-hal"
rev = "f1b44d625e0fbfbba3dc11d53901acdd64026638"
```

PR #655 (2023-06-24): fix #604
Fixes #654 

cc @KarolNi
closed 2023-06-24 by burrbull