stm32-rs/stm32f4xx-hal #679: Constructing TIM DMA Peripherals

BryceBeagle, 2023-08-12:
Hi I'm trying to implement a memory-to-peripheral DMA transfer, and I'm stuck instantiating the `peripheral` for the `Transfer` object

```rust
    let mut transfer = Transfer::init_memory_to_peripheral(
        StreamsTuple::new(dp.DMA2).6,
        CCR::<TIM1, 0>(dp.TIM1),
        None,
        DmaConfig::default()
            .memory_increment(true)
            .fifo_enable(true)
            .fifo_error_interrupt(true)
            .transfer_complete_interrupt(true),
    );
```

Unfortunately, the `CCR` tuple struct holds `T` as a private field:
https://github.com/stm32-rs/stm32f4xx-hal/blob/47404df8d57f6cdee5c85cebb778f688c2f2059e/src/timer.rs#L246-L248

```
error[E0423]: cannot initialize a tuple struct which contains private fields
   --> src/main.rs:63:9
    |
63  |         CCR::<TIM1, 0>(dp.TIM1),
    |         ^^^^^^^^^^^^^^
    |
```

This looks like the same problem as #227, where the solution was to make `T` public, which was done in #237.
https://github.com/stm32-rs/stm32f4xx-hal/commit/be04db0fc49130b15ca1e8985d7bff4a74888aa9#diff-83bb17ad20b71ad74e5fa57169f61d11aa6d64a5a06cee94537a70a3afebbc4bR288

However, in #481, `T` was made private again when `CCR` was moved from `dma/trait.rs` (https://github.com/stm32-rs/stm32f4xx-hal/pull/481/files#diff-83bb17ad20b71ad74e5fa57169f61d11aa6d64a5a06cee94537a70a3afebbc4bL271-L272) to `timer.rs` (https://github.com/stm32-rs/stm32f4xx-hal/pull/481/files#diff-abbd7706dff206239bd03de70e864a2893146421b4877638f17149848491fcd0R218-R220).

Was this an intentional change? I'm not seeing any other way to create an instance of `CCRX` structs

burrbull, 2023-08-13:
I missed it was public. But is it right way?
I don't know what @thalesfragoso planned to do with this, but he don't react a long time.

Do you need only instance constructor or you need access to `T`? Do you want to configure timer manually? Before creating `CCRx`? Would `unsafe fn new(timer: T)` be enough?

In my view `CCRx` should be created some way from `Pwm` (or maybe `PwmChannel`) instance. Or maybe full `Transfer` object. But I never used it this way so can't say what pitfalls can be there.

lowstar, 2025-01-13:
I hit this exact issue just today. Was this already resolved somehow?