stm32-rs/stm32f4xx-hal #681: STM32F405 UART missing impls in v0.16 and v0.17

hacknus, 2023-09-17:
in v0.15 I was able to call `uart.read()` and `uart.bwrite_all(&buf)`, but since v0.16 I get errors even though I have the imports:
```
use stm32f4xx_hal::prelude::*;
use embedded_hal::serial::Read;
```

init code:
```rust
let mut uart = Serial::new(
    dp.USART1,
    (tx_1, rx_1),
    Config::default()
        .baudrate(9600.bps())
        .parity_none()
        .stopbits(StopBits::STOP1),
    &clocks,
)
.unwrap();
```

```rust 
error[E0599]: the method `bwrite_all` exists for mutable reference `&mut Serial<USART1, (Pin<'A', 9>, Pin<'A', 10>)>`, but its trait bounds were not satisfied
   --> src/uart.rs:60:18
    |
60  |             uart.bwrite_all(data);
    |                  ^^^^^^^^^^ method cannot be called on `&mut Serial<USART1, (Pin<'A', 9>, Pin<'A', 10>)>` due to unsatisfied trait bounds
    |
   ::: /Users/linus/.cargo/git/checkouts/stm32f4xx-hal-fe8350cc04cacf3f/858163f/src/serial.rs:121:1
    |
121 | pub struct Serial<USART: CommonPins, WORD = u8> {
    | -----------------------------------------------
    | |
    | doesn't satisfy `_: Default<_>`
    | doesn't satisfy `_: _embedded_hal_blocking_serial_Write<_>`
    |
    = note: the following trait bounds were not satisfied:
            `Serial<stm32f4xx_hal::pac::USART1, (stm32f4xx_hal::gpio::Pin<'A', 9>, stm32f4xx_hal::gpio::Pin<'A', 10>)>: embedded_hal::blocking::serial::write::Default<_>`
            which is required by `Serial<stm32f4xx_hal::pac::USART1, (stm32f4xx_hal::gpio::Pin<'A', 9>, stm32f4xx_hal::gpio::Pin<'A', 10>)>: _embedded_hal_blocking_serial_Write<_>`
            `&mut Serial<stm32f4xx_hal::pac::USART1, (stm32f4xx_hal::gpio::Pin<'A', 9>, stm32f4xx_hal::gpio::Pin<'A', 10>)>: embedded_hal::blocking::serial::write::Default<_>`
            which is required by `&mut Serial<stm32f4xx_hal::pac::USART1, (stm32f4xx_hal::gpio::Pin<'A', 9>, stm32f4xx_hal::gpio::Pin<'A', 10>)>: _embedded_hal_blocking_serial_Write<_>`
```

burrbull, 2023-09-17:
Both traits are implemented.

Can you show your `cargo tree`? Error looks strange for 0.16.

hacknus, 2023-09-17:
Thanks @burrbull for looking into this. The error above is for the most recent github state of `stm32f4xx-hal`. But it looks really similar for 0.16:
```rust 
error[E0599]: the method `bwrite_all` exists for mutable reference `&mut Serial<USART1, (Pin<'A', 9>, Pin<'A', 10>)>`, but its trait bounds were not satisfied
   --> src/uart.rs:60:18
    |
60  |             uart.bwrite_all(data);
    |                  ^^^^^^^^^^ method cannot be called on `&mut Serial<USART1, (Pin<'A', 9>, Pin<'A', 10>)>` due to unsatisfied trait bounds
    |
   ::: /Users/linus/.cargo/registry/src/index.crates.io-6f17d22bba15001f/stm32f4xx-hal-0.16.2/src/serial.rs:121:1
    |
121 | pub struct Serial<USART: CommonPins, WORD = u8> {
    | -----------------------------------------------
    | |
    | doesn't satisfy `_: Default<_>`
    | doesn't satisfy `_: _embedded_hal_blocking_serial_Write<_>`
    |
    = note: the following trait bounds were not satisfied:
            `Serial<stm32f4xx_hal::pac::USART1, (stm32f4xx_hal::gpio::Pin<'A', 9>, stm32f4xx_hal::gpio::Pin<'A', 10>)>: embedded_hal::blocking::serial::write::Default<_>`
            which is required by `Serial<stm32f4xx_hal::pac::USART1, (stm32f4xx_hal::gpio::Pin<'A', 9>, stm32f4xx_hal::gpio::Pin<'A', 10>)>: _embedded_hal_blocking_serial_Write<_>`
            `&mut Serial<stm32f4xx_hal::pac::USART1, (stm32f4xx_hal::gpio::Pin<'A', 9>, stm32f4xx_hal::gpio::Pin<'A', 10>)>: embedded_hal::blocking::serial::write::Default<_>`
            which is required by `&mut Serial<stm32f4xx_hal::pac::USART1, (stm32f4xx_hal::gpio::Pin<'A', 9>, stm32f4xx_hal::gpio::Pin<'A', 10>)>: _embedded_hal_blocking_serial_Write<_>`

```
It's quite a large project. I could provide a stripped down example later on. But here is the result of `cargo tree` with v0.17.
```shell

icelab-elu v0.0.1 (/Users/linus/CLionProjects/icelab-elu)
├── ads122x04 v0.1.0 (https://github.com/hacknus/ads122x04-rs#07a1acfe)
│   ├── embedded-hal v0.2.7
│   │   ├── nb v0.1.3
│   │   │   └── nb v1.1.0
│   │   └── void v1.0.2
│   └── nb v1.1.0
├── arrform v0.1.1
├── cortex-m v0.7.7
│   ├── bare-metal v0.2.5
│   │   [build-dependencies]
│   │   └── rustc_version v0.2.3
│   │       └── semver v0.9.0
│   │           └── semver-parser v0.7.0
│   ├── bitfield v0.13.2
│   ├── critical-section v1.1.2
│   ├── embedded-hal v0.2.7 (*)
│   └── volatile-register v0.2.1
│       └── vcell v0.1.3
├── cortex-m-rt v0.7.3
│   └── cortex-m-rt-macros v0.7.0 (proc-macro)
│       ├── proc-macro2 v1.0.67
│       │   └── unicode-ident v1.0.12
│       ├── quote v1.0.33
│       │   └── proc-macro2 v1.0.67 (*)
│       └── syn v1.0.109
│           ├── proc-macro2 v1.0.67 (*)
│           ├── quote v1.0.33 (*)
│           └── unicode-ident v1.0.12
├── embedded-hal v0.2.7 (*)
├── freertos-rust v0.1.2
├── micromath v2.0.0
├── modular-bitfield v0.11.2
│   ├── modular-bitfield-impl v0.11.2 (proc-macro)
│   │   ├── proc-macro2 v1.0.67 (*)
│   │   ├── quote v1.0.33 (*)
│   │   └── syn v1.0.109 (*)
│   └── static_assertions v1.1.0
├── nb v1.1.0
├── ringbuffer v0.14.2
├── rtt-target v0.3.1
│   ├── cortex-m v0.7.7 (*)
│   └── ufmt-write v0.1.0
├── shared-bus v0.3.0
[... 138 lines cut ...]
│   │   └── vcell v0.1.3
│   ├── synopsys-usb-otg v0.3.2
│   │   ├── cortex-m v0.7.7 (*)
│   │   ├── embedded-hal v0.2.7 (*)
│   │   ├── usb-device v0.2.9
│   │   └── vcell v0.1.3
│   ├── time v0.3.28
│   │   ├── deranged v0.3.8
│   │   └── time-core v0.1.1
│   └── void v1.0.2
├── tmc5160 v0.0.1 (https://github.com/hacknus/tmc5160-rs#a80f4462)
│   ├── embedded-hal v0.2.7 (*)
│   └── modular-bitfield v0.11.2 (*)
├── usb-device v0.2.9
└── usbd-serial v0.1.1
    ├── embedded-hal v0.2.7 (*)
    ├── nb v0.1.3 (*)
    └── usb-device v0.2.9
[build-dependencies]
└── freertos-cargo-build v0.1.1
    ├── cc v1.0.83
    │   └── libc v0.2.148
    └── walkdir v2.4.0
        └── same-file v1.0.6
```

burrbull, 2023-09-17:
Current `Serial` doesn't have `PINS` generic:
```rust
pub struct Serial<USART: CommonPins, WORD = u8> {
    tx: Tx<USART, WORD>,
    rx: Rx<USART, WORD>,
}
```
But you've created old `Serial` which does have: `Serial<USART1, (Pin<'A', 9>, Pin<'A', 10>)>`.
It could be only in one situation - conflicting crates.

burrbull, 2023-09-17:
Maybe Clion issue? Try clean cache and `cargo update`.

hacknus, 2023-09-17:
> Current `Serial` doesn't have `PINS` generic:
>
> ```rust
> pub struct Serial<USART: CommonPins, WORD = u8> {
>     tx: Tx<USART, WORD>,
>     rx: Rx<USART, WORD>,
> }
> ```
>
> But you've created old `Serial` [...]

Perfect, thank you very much. I indeed used the old implementation in another part of the code. Thanks!

closed 2023-09-17 by hacknus