stm32-rs/stm32f4xx-hal #685: I2S master clock not working in transmit mode

JosefUtbult, 2023-10-03:
Hello.

I've been trying to get an I2S transmit stream to work. I've been using the following configuration from the [i2s-audio-out example](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/examples/i2s-audio-out.rs).

```rust
let i2s_transmit = I2s::new(codec.spi2, i2s_transmit_pins, codec.clocks);

// Setup Master device for configurations codec
let i2s_transmit_config = I2sDriverConfig::new_master()
    .receive()
    .standard(Philips)
    .data_format(DataFormat::Data24Channel32)
    .master_clock(true)
    .request_frequency(48_000);

let mut i2s_transmit_driver = I2sDriver::new(i2s_transmit, i2s_transmit_config);
defmt::info!("Actual sample rate is {}", i2s_transmit_driver.sample_rate());

i2s_transmit_driver.enable();
```

Which generates a functioning master clock output on pin C6. The problem though is that this master clock doesn't function when I'm changing it to transmit mode.

```rust
let i2s_transmit = I2s::new(codec.spi2, i2s_transmit_pins, codec.clocks);

// Setup Master device for configurations codec
let i2s_transmit_config = I2sDriverConfig::new_master()
    .transmit()
    .standard(Philips)
    .data_format(DataFormat::Data24Channel32)
    .master_clock(true)
    .request_frequency(48_000);

let mut i2s_transmit_driver = I2sDriver::new(i2s_transmit, i2s_transmit_config);
defmt::info!("Actual sample rate is {}", i2s_transmit_driver.sample_rate());

i2s_transmit_driver.enable();
```

Is this a problem with how the STM32 works, or is it something thats missing in the HAL? Thanks in advance

burrbull, 2023-10-03:
cc @YruamaLairba

JosefUtbult, 2023-10-04:
Good call. The problem was in that I didn't send any data to start with. Now the problem lays in that the codec I'm using needs a master clock signal in order to be accessable over i2c for configuration. I solved it by configuring it as a i2s receive with a master clock, wrote the configuration over i2c and then reconfigured it to a i2s transmit. Thanks for the help

YruamaLairba, 2023-10-04:
@JosefUtbult What codec do you use ? Are you sure to not confuse I2s Master clock with I2c Master clock ?