stm32-rs/stm32f4xx-hal #686: embedded-hal  1.0.0-rc.1

pdgilbert, 2023-10-06:
Any timeline for updating `[dependencies.embedded-hal-one]` to `1.0.0-rc.1` ? Or is there a branch that already has it?

rursprung, 2023-12-05:
the update to 1.0.0-rc.1 has already been released with 0.18.0.

however, the update to e-h 1.0.0-rc.2 is not yet released (though @burrbull has done it with commit a68a3a1) - it'd be great to get a release with this!

pdgilbert, 2023-12-14:
Using a recent git version with @burrbull 's commit I am now running several examples with `e-h 1.0.0-rc.2` at https://github.com/pdgilbert/rust-integration-testing/actions. See the `eh-1-rc2` workflows. The biggest changes were needing to update from `rtic v1` to `rtic v2` and needing delay traits  from `DelayNs`. Most examples are compiling but note that I have not yet done any hardware testing.

While the `hal` and my code are using `e-h 1.0.0-rc.2`, the device crates are all using `embedded-hal v0.2.7`. The `tree` is pretty ugly. I hope the quality of this magic dual support in `stm32f4xx-hal` does not make device crate maintainers lazy and slow to move forward.

rursprung, 2023-12-16:
FYI: i've created #715 to update to `1.0.0-rc.3`.

> I hope the quality of this magic dual support in `stm32f4xx-hal` does not make device crate maintainers lazy and slow to move forward.

i _presume_ that after 1.0.0 has been released there'll be a period where HALs will support 0.2.x and 1.x in parallel to offer a transition period (without being stuck on an old HAL) while drivers will update to supporting purely 1.x. i _hope_ that this period will be short and that HALs will drop e-h 0.x support soon after (to create some incentive for everyone to _really_ update). theoretically HALs could also immediately drop e-h 0.x support and say "if you need 0.x just stay on the last HAL version supporting it and upgrade once you're ready".

rursprung, 2023-12-29:
i _think_ that this issue can be closed as the HAL already implements the latest available e-h version (`1.0.0-rc.3` at the moment, which is newer than the requested `1.0.0-rc.1`).

i've opened #721 since i noticed that the UART related traits are not yet implemented.

pdgilbert, 2023-12-29:
Yes, this issue has expired. Looking forward to actual release of eh-1.0.0 very soon.

closed 2023-12-29 by pdgilbert