stm32-rs/stm32f4xx-hal #690: how does "usb_fs" and "usb_hs" work together?

xzzWZY, 2023-11-03:
in Cargo.toml, I try to include both "usb_fs" and "usb_hs" like 
```stm32f4xx-hal = { version = "0.17.1", features = [
  "stm32f405",
  "rtic",
  "rtic-monotonic",
  "usb_fs",
  "usb_hs",
  "synopsys-usb-otg",
  "i2s",
  "sdio-host",
] }```

but there's error saying
```
error[E0252]: the name `OTG_GLOBAL` is defined multiple times
  --> {...}/synopsys-usb-otg-0.3.2/src/ral/mod.rs:18:13
   |
16 |     pub use super::stm32f429::otg_fs_global::OTG_FS_GLOBAL as OTG_GLOBAL;
   |             ------------------------------------------------------------ previous import of the module `OTG_GLOBAL` here
17 |     #[cfg(feature = "hs")]
18 |     pub use super::stm32f429::otg_hs_global::OTG_HS_GLOBAL as OTG_GLOBAL;
   |             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ `OTG_GLOBAL` reimported here
   |
   = note: `OTG_GLOBAL` must be defined only once in the type namespace of this module
```

what may cause this issue? how to use both 'usb_fs' and 'usb_hs' together?

paulyoung, 2023-11-17:
```toml
[features]
fs = ["stm32f4xx-hal/usb_fs"]
hs = ["stm32f4xx-hal/usb_hs"]

[dependencies]
stm32f4xx-hal = { version = "0.17.1", features = [
  "stm32f405",
  "rtic",
  "rtic-monotonic",
  "synopsys-usb-otg",
  "i2s",
  "sdio-host",
] }
```

Then build with `--features fs` or `--features hs`

paulyoung, 2023-11-17:
Unless you mean this? https://github.com/stm32-rs/synopsys-usb-otg/issues/12