stm32-rs/stm32f4xx-hal #721: implement `embedded-io` traits

rursprung, 2023-12-29:
with `embedded-hal` v1 the UART related traits have been removed in favour of [`embedded-io`](https://crates.io/crates/embedded-io). thus these traits should be implemented by this HAL so that the replacement for the old e-h 0.2 traits is available.

for avr-hal i've done an attempt at this, maybe that helps as a reference: https://github.com/Rahix/avr-hal/pull/484 (i don't have an STM32 at hand, so i can't provide it here, sorry)

pdgilbert, 2024-02-07:
I am working on an example that reads from one serial port and writes to another using `Read` and `Write` traits from `embedded-io-0.6.1`  (and using `embedded-hal-1.0.0`). The code further below compiles using  @techmccat 's fork of `stm32g4xx-hal` and branch `eh-v1.0` of `stm32h7xx-hal`. Compiling with `stm32f4xx-hal` gives these errors:

<details>
<summary>Click to expand compile errors</summary>

```
$ cargo build --no-default-features --target thumbv7em-none-eabihf --features stm32f411,stm32f4xx --example echo_by_char
    Updating crates.io index
    Updating git repository `https://github.com/techmccat/stm32g4xx-hal`
    Updating git repository `https://github.com/stm32-rs/stm32h7xx-hal`
   Compiling proc-macro2 v1.0.78
   Compiling unicode-ident v1.0.12
   Compiling semver-parser v0.7.0
   Compiling cortex-m v0.7.7
   Compiling nb v1.1.0
   Compiling semver v1.0.21
   Compiling vcell v0.1.3
   Compiling cortex-m-rt v0.7.3
   Compiling syn v1.0.109
   Compiling nb v0.1.3
   Compiling void v1.0.2
   Compiling volatile-register v0.2.2
   Compiling embedded-hal v0.2.7
   Compiling critical-section v1.1.2
   Compiling semver v0.9.0
   Compiling autocfg v1.1.0
   Compiling bitfield v0.13.2
   Compiling rustc_version v0.2.3
   Compiling byteorder v1.5.0
   Compiling az v1.2.1
   Compiling gcd v2.3.0
   Compiling stm32f4 v0.15.1
   Compiling rustc_version v0.4.0
   Compiling bare-metal v0.2.5
   Compiling stable_deref_trait v1.2.0
   Compiling powerfmt v0.2.0
   Compiling num-traits v0.2.17
   Compiling cortex-m-semihosting v0.5.0
   Compiling deranged v0.3.11
   Compiling quote v1.0.35
   Compiling syn v2.0.48
   Compiling heapless v0.7.17
   Compiling fugit v0.3.7
   Compiling stm32f4xx-hal v0.20.0
   Compiling num-conv v0.1.0
   Compiling time-core v0.1.2
   Compiling embedded-hal v1.0.0
   Compiling litrs v0.4.1
   Compiling bare-metal v1.0.0
   Compiling embedded-hal-nb v1.0.0
   Compiling time v0.3.34
   Compiling fugit-timer v0.1.3
   Compiling embedded-dma v0.2.0
   Compiling hash32 v0.2.1
   Compiling rand_core v0.6.4
   Compiling float-cmp v0.9.0
   Compiling embedded-graphics-core v0.4.0
   Compiling micromath v2.1.0
   Compiling embedded-storage v0.3.1
   Compiling panic-semihosting v0.6.0
   Compiling panic-halt v0.2.0
   Compiling embedded-io v0.6.1
   Compiling document-features v0.2.8
   Compiling embedded-graphics v0.8.1
   Compiling enumflags2_derive v0.7.8
   Compiling cortex-m-rt-macros v0.7.0
   Compiling enumflags2 v0.7.8
   Compiling echo v0.0.1 (echo)
error[E0308]: mismatched types
   --> examples/echo_by_char.rs:131:15
[... 274 lines cut ...]
embedded-graphics = ">=0.7"
heapless = "0.7"
cortex-m-rt = ">=0.7.0"
cortex-m-semihosting = { version = "0.5.0" }
panic-halt           = { version = ">=0.2.0" }
panic-semihosting    = { version = ">=0.6.0" }

[features]
stm32f4xx = ["stm32f4xx-hal" ] 
stm32g4xx = ["stm32g4xx-hal" ]
stm32h7xx = ["stm32h7xx-hal" ]

stm32f401     = ["stm32f4xx-hal/stm32f401"  ] 
stm32f411     = ["stm32f4xx-hal/stm32f411"  ] 
stm32g473     = ["stm32g4xx-hal/stm32g473"  ]
stm32g474     = ["stm32g4xx-hal/stm32g474"  ]
stm32h742     = ["stm32h7xx-hal/stm32h742"  ]

```

</details>

These are also available in a package structure at https://github.com/pdgilbert/echo .

This example is derived from an example that worked with `embedded-hal-0.2.7` by reading and writing individual bytes. As pointed out by @richardion (see https://github.com/stm32-rs/stm32h7xx-hal/issues/474#issuecomment-1883838622 ), there are no methods for byte wise serial operations in `embedded-io`. Thus the read/write buffer in the example can now be more than a single byte.  I am not sure the example code will accomplish this properly, so comments on the code would also be appreciated.

pdgilbert, 2024-02-08:
Also tried with the `stm32f4xx_hal` `io` branch for @burrbull 's pull request https://github.com/stm32-rs/stm32f4xx-hal/pull/725. The error message suggests to disambiguate the method. So changing, for example,  `tx1.write(b"\r\nconsole connect check.\r\n").ok();` to `embedded_io::Write::write(&mut tx1, b"\r\nconsole connect check.\r\n").ok();` the writes do not cause errors. It does not address `Read`. Attempting read with `let _len = embedded_io::Read::read(&mut rx1, &mut buffer).ok();` to disambiguate gives

<details>
<summary>Click to expand compile errors</summary>

```
$ cargo build  --no-default-features --target $TARGET --features $MCU,$HAL --example echo_by_char
   Compiling echo v0.0.1 (echo)
warning: unused imports: `Read`, `Write`
  --> examples/echo_by_char.rs:23:19
   |
23 | use embedded_io::{Read, Write};
   |                   ^^^^  ^^^^^
   |
   = note: `#[warn(unused_imports)]` on by default

error[E0277]: the trait bound `stm32f4xx_hal::serial::Rx<stm32f4xx_hal::pac::USART1>: embedded_io::Read` is not satisfied
   --> examples/echo_by_char.rs:144:44
    |
144 |         let _len = embedded_io::Read::read(&mut rx1, &mut buffer).ok();
    |                    ----------------------- ^^^^^^^^ the trait `embedded_io::Read` is not implemented for `stm32f4xx_hal::serial::Rx<stm32f4xx_hal::pac::USART1>`
    |                    |
    |                    required by a bound introduced by this call
    |
    = help: the following other types implement trait `embedded_io::Read`:
              &[u8]
              &mut T

For more information about this error, try `rustc --explain E0277`.
warning: `echo` (example "echo_by_char") generated 1 warning
error: could not compile `echo` (example "echo_by_char") due to 1 previous error; 1 warning emitted

```

</details>