stm32-rs/stm32f4xx-hal #722: sandwiched eh versions failure

pdgilbert, 2024-01-05:
I have been testing `embedded-hal-1.0.0-rc3` with relatively good results in many examples, using `stm32f4xx_hal` and `stm32h7xx_hal`. Of course many of my examples use crates which do not have a `embedded-hal-1.0.0-rc3` version. Thus, they fail with `stm32h7xx_hal` because it does not have the magic dual `embedded-hal` version suport. (OTOH, that highlights the device crates that do not yet support `embedded-hal-1.0.0`.)

One problem I have encountered with  `stm32f4xx_hal`  is when `shared-bus` with `embedded-hal = "0.2.3"` is sandwiched between my `embedded-hal-1.0.0` code and an `embedded-hal-1.0.0-rc3` version of the `ssd1306` display driver. I use this driver in several examples.  From details below I hope someone can confirm if I am missing something simple.  Also, if anyone is aware of a `shared-bus` with `embedded-hal = "1.0.0-rc3"` please point me to it. (I have tried @TeyKey1 's fork at https://github.com/TeyKey1/shared-bus"  but I don't think it is working yet - please correct me if I am wrong.)

Below is code simplified from an example that does not necessarily need `shared-bus`. The code works when `shared-bus` is omitted and fails when it is included. The error is

<details>
<summary>Click to expand error messages</summary>

```
error[E0277]: the trait bound `I2cProxy<'static, cortex_m::interrupt::Mutex<RefCell<stm32f4xx_hal::i2c::I2c<I2C1>>>>: embedded_hal::i2c::I2c` is not satisfied
   --> examples/zzdisplay_stuff_rtic2.rs:194:25
    |
194 |            show_display(cx.local.display);
    |            ------------ ^^^^^^^^^^^^^^^^ the trait `embedded_hal::i2c::I2c` is not implemented for `I2cProxy<'static, cortex_m::interrupt::Mutex<RefCell<stm32f4xx_hal::i2c::I2c<I2C1>>>>`
    |            |
    |            required by a bound introduced by this call
    |
    = help: the following other types implement trait `embedded_hal::i2c::I2c<A>`:
              <I2cStub as embedded_hal::i2c::I2c>
              <stm32f4xx_hal::i2c::I2c<I2C> as embedded_hal::i2c::I2c>
              <&mut T as embedded_hal::i2c::I2c<A>>
    = note: required for `I2CInterface<I2cProxy<'static, Mutex<RefCell<I2c<I2C1>>>>>` to implement `ssd1306::prelude::WriteOnlyDataCommand`
    = note: the full type name has been written to '/home/paul/githubClones/rust-integration-testing/dev-testing/target/thumbv7em-none-eabihf/debug/examples/zzdisplay_stuff_rtic2-a07c81e330eef225.long-type-4472187899878778957.txt'
note: required by a bound in `show_display`
   --> examples/zzdisplay_stuff_rtic2.rs:107:33
    |
106 |     fn show_display<S>(
    |        ------------ required by a bound in this function
107 |         disp: &mut Ssd1306<impl WriteOnlyDataCommand, S, BufferedGraphicsMode<S>>,
    |                                 ^^^^^^^^^^^^^^^^^^^^ required by this bound in `show_display`

error[E0599]: the method `flush` exists for mutable reference `&mut Ssd1306<I2CInterface<I2cProxy<'static, Mutex<...>>>, ..., ...>`, but its trait bounds were not satisfied
   --> examples/zzdisplay_stuff_rtic2.rs:211:29
    |
211 |            cx.local.display.flush().unwrap();
    |                             ^^^^^ method cannot be called due to unsatisfied trait bounds
    |
   ::: /home/paul/.cargo/git/checkouts/display-interface-671cffc841c04cf5/0ed9dc9/i2c/src/lib.rs:10:1
    |
10  | pub struct I2CInterface<I2C> {
    | ---------------------------- doesn't satisfy `_: WriteOnlyDataCommand`
    |
    = note: the full type name has been written to '/home/paul/githubClones/rust-integration-testing/dev-testing/target/thumbv7em-none-eabihf/debug/examples/zzdisplay_stuff_rtic2-a07c81e330eef225.long-type-10539551830589570764.txt'
    = note: the following trait bounds were not satisfied:
            `I2CInterface<I2cProxy<'static, cortex_m::interrupt::Mutex<RefCell<stm32f4xx_hal::i2c::I2c<I2C1>>>>>: ssd1306::prelude::WriteOnlyDataCommand`

error[E0277]: the trait bound `I2cProxy<'_, cortex_m::interrupt::Mutex<RefCell<stm32f4xx_hal::i2c::I2c<I2C1>>>>: embedded_hal::i2c::I2c` is not satisfied
   --> examples/zzdisplay_stuff_rtic2.rs:152:49
    |
152 |        let interface = I2CDisplayInterface::new(manager.acquire_i2c());
    |                        ------------------------ ^^^^^^^^^^^^^^^^^^^^^ the trait `embedded_hal::i2c::I2c` is not implemented for `I2cProxy<'_, cortex_m::interrupt::Mutex<RefCell<stm32f4xx_hal::i2c::I2c<I2C1>>>>`
    |                        |
    |                        required by a bound introduced by this call
    |
    = help: the following other types implement trait `embedded_hal::i2c::I2c<A>`:
              <I2cStub as embedded_hal::i2c::I2c>
              <stm32f4xx_hal::i2c::I2c<I2C> as embedded_hal::i2c::I2c>
              <&mut T as embedded_hal::i2c::I2c<A>>
note: required by a bound in `I2CDisplayInterface::new`
   --> /home/paul/.cargo/git/checkouts/ssd1306-407b1a92e31533f1/c019ef1/src/i2c_interface.rs:15:12
    |
13  |     pub fn new<I>(i2c: I) -> I2CInterface<I>
    |            --- required by a bound in this associated function
14  |     where
15  |         I: embedded_hal::i2c::I2c,
    |            ^^^^^^^^^^^^^^^^^^^^^^ required by this bound in `I2CDisplayInterface::new`

error[E0599]: the method `init` exists for struct `Ssd1306<I2CInterface<I2cProxy<'_, Mutex<RefCell<...>>>>, ..., ...>`, but its trait bounds were not satisfied
   --> examples/zzdisplay_stuff_rtic2.rs:161:16
[... 232 lines cut ...]
           cx.local.display.clear_buffer();

           write!(lines[1], "stuff").unwrap();

           for i in 0..lines.len() {
              // start from 0 requires that the top is used for font baseline
              Text::with_baseline(
                  &lines[i],
                  Point::new(0, i as i32 * 12), //with font 6x10, 12 = 10 high + 2 space
                  //cx.local.text_style,
                  text_style,
                  Baseline::Top,
              ).draw(cx.local.display).unwrap();
           }

           cx.local.display.flush().unwrap();

           Systick::delay(READ_INTERVAL.secs()).await;
       }        
     }
   }
``` 
</details>

Results from my testing are at https://github.com/pdgilbert/rust-integration-testing/actions in the `eh-1-rc3` workflow runs. The code above is extracted  from example `display_stuff_rtic`. A different problem regarding `serial` is reported at https://github.com/stm32-rs/stm32h7xx-hal/issues/474#issuecomment-1879242912

TeyKey1, 2024-01-07:
Hi @pdgilbert my fork has updated to rc3 but I haven't tested it yet so it's wuite possible that it does not work as expected. I'll likely stay on rc2 in my libraries until 1.0 lands so I won't adjust the fork anymore.

pdgilbert, 2024-01-07:
Ok, thanks @TeyKey1  for the confirmation.  I can wait, hopefully it won't be long.

rursprung, 2024-01-09:
see [the migration guide](https://github.com/rust-embedded/embedded-hal/blob/master/docs/migrating-from-0.2-to-1.0.md), `shared-bus` shouldn't be used anymore. you can now use [`embedded-hal-bus`](https://crates.io/crates/embedded-hal-bus).

pdgilbert, 2024-01-13:
Thanks for point this out @rursprung . I am not having much luck with the `embedded-hal-bus` formulation but that is a different problem. I will close this as the sandwich issue is probably not too important anymore.

closed 2024-01-13 by pdgilbert