stm32-rs/stm32f4xx-hal #740: Blinky Example Does Not Work With STM32-F4VE

joeyame, 2024-02-09:
Hi there, I just started setting up my environment to try developing some embedded software for this board: https://stm32-base.org/boards/STM32F407VET6-STM32-F4VE-V2.0.html#User-LED-1

Here are the steps I took:
- Set up the build targets
- Copy the blinky.rs file
- Change GPIOC and pc13 to GPIOA and pa6
- `cargo build --release`
- Rename binary output in `target/thumbv7em-none-eabihf/release` to have a `.bin` extension
- Load the binary file onto the microcontroller via ST Link to memory address 0x08000000. (download & verification completes successfully)
- thumbv7em-none-eabi & thumbv7em-none-eabihf targets

This all results in no blinking. I have even tried using other GPIO pins and holding an LED up to them, but there is no activity.

Tomorrow I'll test if I can get an old program in C++ to run on it to see if it is truly Rust + this framework or if there is something wrong with my board. Either way, am I missing any obvious steps?

burrbull, 2024-02-09:
> * Rename binary output in `target/thumbv7em-none-eabihf/release` to have a `.bin` extension

This is soooo weird.

joeyame, 2024-02-09:
> > * Rename binary output in `target/thumbv7em-none-eabihf/release` to have a `.bin` extension
> 
> This is soooo weird.

Yeah I know, but st-link requires the files to have a bin extension

burrbull, 2024-02-09:
If you really want to get `.bin` you need to use `cargo-binutils` (wrapper for llvm-tools) for this.

But you do not need a `.bin`. Just use `probe-rs` to load firmware with your st-link.

And read [README](https://github.com/stm32-rs/stm32f4xx-hal#fast-start)

P.S. `Issues` is not a blog.