stm32-rs/stm32f4xx-hal #748: Release Rx, Tx resource back to Pin

lgjonathan, 2024-03-11:
Hi,
I have a use case that I need to send a pulse with UART before communicating.
So my approach was to call `set_high`, `set_low` and `delay` to send a pulse.

But once the USART consumes the pin, I couldn't release the pin back to output pin.
From what I see, `Serial::new()` and `Serial::release()` is not bidirectional because Tx takes the ownership of the underlying pin.

I was wondering if there's a way to release `Tx` back to `Pin`?
(so I can call into_push_pull_output)

Thanks in advance

burrbull, 2024-03-11:
You just need to unwrap `Tx` pin enum. https://docs.rs/stm32f4xx-hal/latest/stm32f4xx_hal/gpio/alt/usart1/enum.Tx.html#impl-TryFrom%3CTx%3COtype%3E%3E-for-Pin%3C'A',+9,+MODE%3E

```rust
let (usart, (txpin, rxpin)) = serial.release();
let txpin: gpio::PA8<Output> = txpin.try_into().unwrap();
```

lgjonathan, 2024-03-11:
Works! Much thanks

lgjonathan, 2024-03-11:
Works! Much thanks

closed 2024-03-11 by lgjonathan