stm32-rs/stm32f4xx-hal #750: Implement DataPins8 and extend FSMC abstractions to support 8-bit buses

ellenhp, 2024-04-20:
I don't see an issue tracking support for 8-bit FSMC buses. I forked the HAL for one of my projects that requires an 8-bit FSMC bus, but it ended up being a little complicated to add support for this. I ended up breaking DataPins16 which is fine for my fork, but since I'd like to avoid maintaining a forked HAL I thought I'd open an issue to track this.

burrbull, 2024-04-20:
What exactly have you changed? Where can I look your fork?

ellenhp, 2024-04-20:
I'm vendoring it right now in another project, I'll make the same changes in a proper fork. I'll need about 30 minutes for that, I'll comment again when I have something ready to look at.

ellenhp, 2024-04-20:
These are the changes I made.

https://github.com/ellenhp/stm32f4xx-hal/commit/8e1db40324eb98132eaaa7df6a501e38fa2e78c8

Obviously they can't be merged like that, but it could serve as a starting point.

burrbull, 2024-04-20:
In first look you need additional type parameter (`u8` or `u16`) for `Lcd`, `FsmcLcd` structs and `DataPins`, `LcdPins` traits.

PR #751 (2024-04-27): support FMC u8 data width
Closes #750.

cc @ellenhp try
closed 2024-04-27 by burrbull