stm32-rs/stm32f4xx-hal #752: Implement DMA for DualI2s.

ProjectRobal, 2024-04-25:
Hello, I want to read audio data from ADC using I2S of STM32F411 microcontroller. I want to use DMA to do so. But I have no idea how to configure it. I have created a topic on the rust embedded forum, searching for a solution: https://users.rust-lang.org/t/dma-configuration-for-duali2s/110411 
And one of the answers suggests that I cannot use DMA with DualI2s because of HAL oversight.
Is that true or am I missing something?

burrbull, 2024-04-27:
I don't know who can help you with this. Maybe @YruamaLairba ?

YruamaLairba, 2024-04-28:
> And one of the answers suggests that I cannot use DMA with DualI2s because of HAL oversight. Is that true or am I missing something?

it's unfortunately true, it's currently impossible to use DMA with "DualI2s" without bypassing the HAL and doing dirty things. You can't create a "Transfer" around a "DualI2s" because "Transfer" system is build with the idea that one driver hold one peripheral, which is not the case with "DualI2s". 

And, In my opinion, in many situation, "Transfer" system is not appropriate. For example it's impossible to share a device across interruption because Transfer need ownership of the device