stm32-rs/stm32f4xx-hal #808: NO can() found  in CAN1

Admsher, 2024-10-08:
I am trying to use the cansend.rs(in the example) for stm32ff407vgtx board but I am stuck with the error which says:
35 |         let can = dp.CAN1.can();
   |                           ^^^ method not found in `CAN1`

burrbull, 2024-10-09:
Have you enabled `can` feature?
Are traits from `prelude` imported?

Admsher, 2024-10-09:
Yes I found that out and did it but now bxcan gives the issue of incorrect variable parsed int it's builder function, it asks for something called 'Instance' type and rejects the 'Pin' type value which is given in the example

Admsher, 2024-10-10:
issue:
`[E0277] the trait bound `stm32f4xx_hal::can::Can<CAN2>: bxcan::Instance` is not satisfied. 
[Note] the trait `bxcan::Instance` is not implemented for `stm32f4xx_hal::can::Can<CAN2>``

Cargo.toml:
`[package]
name = "stm32-example"
version = "0.1.0"
authors = ["asdgs"]
edition = "2021"

[dependencies]
cortex-m = "0.7"
cortex-m-rt = "0.7"
panic-probe = { version = "0.3.1", features = ["defmt"] }
panic-halt = "0.2.0"
rtt-target = "0.5.0"
embedded-time = "0.12.1"
libm = "0.2.8"
embedded-timers = "0.3.0"
embedded-hal = "1.0.0"
debouncr = "0.2.2"
nb = "1"
bxcan = "0.8.0"

[dependencies.stm32f4xx-hal]
version = "0.22.0"
features = ["stm32f407", "defmt"]

[profile.dev.package."*"]
opt-level = "s"

[profile.release]
codegen-units = 1
incremental = false
debug = true
lto = true
opt-level = "s"

[dev-dependencies]
panic-halt = "0.2.0"
`

burrbull, 2024-10-10:
I finally understood.

Different `bxcan` versions. `f4xx-hal` depends on `0.7`

Admsher, 2024-10-10:
Oh is that so, its funny cause I never got a warning for this version mismatch, i will try this and update on this thread

Admsher, 2024-10-11:
Thats it we need to use bxcan 0.7.0 version and now it works

closed 2024-10-11 by Admsher