stm32-rs/stm32f4xx-hal #810: When configuring the timer in PWM input mode, it is not possible to use the free channels as PWM output

faunel, 2024-10-11:
When using the timer in PWM input mode, channels 1 and 2 will be used as inputs.
However, channels 3 and 4 remain free, but they cannot be used as PWM output.

Previously, in version 0.21.0, this was not possible because the timer was moved to the ownership of the `pwm_hz` method.
```rust
let tim3 = Timer::new(dp.TIM3, &clocks);
tim3.pwm_hz(10.kHz());

// tim3 move to pwm_hz
// So, it's not possible to do this
tim3.pwm_input(1.kHz(), pwm_reader_ch1);
```

I thought that with the new version 0.22.0, where the timer is split into a PWM manager and channels, this would be possible.
But it still doesn't work.
```rust
let pwm_reader_ch1: Pin<'B', 4> = gpiob.pb4.into_pull_down_input();
let tim3 = Timer::new(dp.TIM3, &clocks);
let (mut pwm_mngr, (_ch1, _ch2, ch3, ch4)) = tim3.pwm_hz(10.kHz());
let mut ch3 = ch3.with(gpiob.pb0);
ch3.enable();
ch3.set_duty(50);

// Error: cannot move out of dereference of `PwmHzManager<stm32f4xx_hal::pac::TIM3>`
let monitor: timer::PwmInput<TIM3> = pwm_mngr.pwm_input(1.kHz(), pwm_reader_ch1);
```

burrbull, 2024-10-11:
> When using the timer in PWM input mode, channels 1 and 2 will be used as inputs.
> However, channels 3 and 4 remain free, but they cannot be used as PWM output.

I even did not know something like this is possible. Input and output PWM on one timer.

faunel, 2024-10-11:
Yes, in the STM32 microcontroller, it is possible to configure the same timer to work simultaneously as both an input (PWM input) and an output (PWM output), using different channels for each mode. This allows, for example, measuring an incoming PWM signal on one channel and generating an outgoing PWM signal on another channel of the same timer.

[image]

faunel, 2024-10-11:
However, the functionality of PWM output will be somewhat limited. The timer has a shared clock generator, so the frequencies of PWM input and PWM output will depend on the same base clock signal.

You can configure different parameters for each channel, such as the duty cycle and other settings, but the main PWM frequency will be shared across all channels of a single timer.

Therefore, I think it’s not worth spending time adding such functionality, as it will be quite specific and limited. I just thought I was doing something wrong, but it turns out this is not implemented.

burrbull, 2024-10-17:
I like your idea. But I need some time to think over the details.