stm32-rs/stm32f4xx-hal #811: the trait bound `stm32f4xx_hal::gpio::alt::spi1::Sck: From<stm32f4xx_hal::gpio::Pin<'B', 13, stm32f4xx_hal::gpio::Alternate<_>>>` is not satisfied

Admsher, 2024-10-11:
I am trying to use  spi code for the board stm32f407vgt and i am getting this erro in lin:
  let gpiob = dp.GPIOB.split();
    let sclk = gpiob.pb13.into_alternate();
    let mosi = gpiob.pb15.into_push_pull_output(); 

let mut spi = dp.SPI1.spi(
        (sclk, NoMiso {}, mosi),
        Mode {
            polarity: Polarity::IdleLow,
            phase: Phase::CaptureOnFirstTransition,
        },
        2.MHz(),
        &clocks,
    );

the error is:
the trait bound `stm32f4xx_hal::gpio::alt::spi1::Sck: From<stm32f4xx_hal::gpio::Pin<'B', 13, stm32f4xx_hal::gpio::Alternate<_>>>` is not satisfied
this is for all the pins 

Cargo.toml:
            [package]
            name = "stm32-example"
            version = "0.1.0"
            authors = ["asdgs"]
            edition = "2021"

            [dependencies]
            cortex-m = "0.7"
            cortex-m-rt = "0.7"
            panic-probe = { version = "0.3.1", features = ["defmt"] }
            panic-halt = "0.2.0"
            rtt-target = "0.5.0"
            embedded-time = "0.12.1"
            libm = "0.2.8"
            embedded-timers = "0.3.0"
            embedded-hal = "1.0.0"
            debouncr = "0.2.2"
            nb = "1"
            bxcan = "0.7.0"

            [dependencies.stm32f4xx-hal]
            version = "0.22.0"
            features = ["stm32f407", "defmt","can"]

            # Set the default for dependencies.
            [profile.dev.package."*"]
            opt-level = "s"

            [profile.release]
            codegen-units = 1
            incremental = false
            debug = true
            lto = true
            opt-level = "s"

            [dev-dependencies]
            panic-halt = "0.2.0"

burrbull, 2024-10-11:
`let mut spi = dp.SPI1.spi(`

there should be `SPI2`

Admsher, 2024-10-12:
https://github.com/stm32-rs/stm32f4xx-hal/blob/master/examples/spi-dma.rs i amusing this code directly, it does say it has spi2

burrbull, 2024-10-12:
You try to use **SPI1** with SPI2 pins. Look again at **your** code.

Admsher, 2024-10-12:
oh i see the error ,  i should be using pa5,pa6 and pa7 , thanks for the help :)

closed 2024-10-12 by Admsher