stm32-rs/stm32f4xx-hal #826: st7789-lcd example not working on stm32f412 disco 1 board.

kareled, 2024-12-02:
I'm trying to run st7789-lcd example on the stm32f412 disco 1 board, but the only visible effect is that display switch on, underlying lightining is on, but otherwise nothing is shown on it.

burrbull, 2024-12-02:
Unfortunately many examples are not tested since adding in the repo.
Could you try older version of hal/example.

If https://github.com/stm32-rs/stm32f4xx-hal/tree/c2bf09a479f21f60e2f552141b6cc64f44ffa323 works for you we could try to find the cause of issue.

kareled, 2024-12-02:
Hmm, needed to comment out one probably too old code unsafe line in src/spi.rs, but then it really starts working!

Unfortunately ST7789 driver as it is in 0.7.0 version is not so good for stm32f412 disco as it's delays in reset function are not well adjusted. Based on stm example at least I've hacked my own copy to include following diff:
```
diff --git a/src/lib.rs b/src/lib.rs
index d010346..643c21f 100644
--- a/src/lib.rs
+++ b/src/lib.rs
@@ -160,12 +160,26 @@ where
     ///
     pub fn hard_reset(&mut self, delay_source: &mut impl DelayUs<u32>) -> Result<(), Error<PinE>> {
         if let Some(rst) = self.rst.as_mut() {
+            // adjusted for STM32F412g-disco1 board
+/*
+      /* Apply hardware reset according to procedure indicated in FRD154BP2901 documentation */
+  HAL_GPIO_WritePin(LCD_RESET_GPIO_PORT, LCD_RESET_PIN, GPIO_PIN_RESET);
+  HAL_Delay(5);   /* Reset signal asserted during 5ms  */
+  HAL_GPIO_WritePin(LCD_RESET_GPIO_PORT, LCD_RESET_PIN, GPIO_PIN_SET);
+  HAL_Delay(10);  /* Reset signal released during 10ms */
+  HAL_GPIO_WritePin(LCD_RESET_GPIO_PORT, LCD_RESET_PIN, GPIO_PIN_RESET);
+  HAL_Delay(20);  /* Reset signal asserted during 20ms */
+  HAL_GPIO_WritePin(LCD_RESET_GPIO_PORT, LCD_RESET_PIN, GPIO_PIN_SET);
+  HAL_Delay(10);  /* Reset signal released during 10ms */
+*/
+            rst.set_low().map_err(Error::Pin)?;
+            delay_source.delay_us(5_000);
             rst.set_high().map_err(Error::Pin)?;
-            delay_source.delay_us(10); // ensure the pin change will get registered
+            delay_source.delay_us(10_000); // ensure the pin change will get registered
             rst.set_low().map_err(Error::Pin)?;
-            delay_source.delay_us(10); // ensure the pin change will get registered
+            delay_source.delay_us(20_000); // ensure the pin change will get registered
             rst.set_high().map_err(Error::Pin)?;
-            delay_source.delay_us(10); // ensure the pin change will get registered
+            delay_source.delay_us(10_000); // ensure the pin change will get registered
         }

         Ok(())
```
but never pushed that upstream. So old example code with its reset is probably the right here. Another issue is display interface, but it's already nearly a year so I somewhat do not remember clearly....