stm32-rs/stm32f4xx-hal #827: STM32F401: UART1 does not work send or recieve characters.

saqibjaved977, 2024-12-26:
Hi, 

I have following code from the serial example. I am using UART1 on stm32f401 with pins PA9 and PA10. Baudraet is 9600, I have a bluetooth module HC-05 connected. 
Rx<->Tx PA9
Tx<->Rx PA10

I connect bluetooth module to my phone and use an app to send and read data, but I do not recieve it and cant read it back as well. 

Anyone who can help me with it?

`#![no_main]
#![no_std]

use panic_halt as _;

use stm32f4xx_hal as hal;

use crate::hal::{block, pac, prelude::*, serial::config::Config};
use cortex_m_rt::entry;

use core::fmt::Write;

#[entry]
fn main() -> ! {
    let dp = pac::Peripherals::take().unwrap();
    let cp = cortex_m::peripheral::Peripherals::take().unwrap();

    let gpioa = dp.GPIOA.split();
    //let gpiob = dp.GPIOB.split();
    let mut led = gpioa.pa5.into_push_pull_output();

    let rcc = dp.RCC.constrain();

    let clocks = rcc.cfgr.use_hse(8.MHz()).freeze();

    let mut delay = cp.SYST.delay(&clocks);

    // define RX/TX pins
    let tx_pin = gpioa.pa9.into_alternate();
    let rx_pin = gpioa.pa10.into_alternate();

    // configure serial
    let serial = dp
        .USART1
        .serial(
            (tx_pin, rx_pin),
            Config::default().baudrate(9600.bps()),
            &clocks,
        )
        .unwrap();

    let (mut tx, mut rx) = serial.split();

    loop {
        // Receive what we just sent
        let received: u16 = rx.read().unwrap();

        // Update LEDs to display what was received
        if received == 1 {
            led.set_high();
        } else {
            led.set_low();
        }
        tx.write(received).unwrap();

        delay.delay_ms(10);
    }
}
`

burrbull, 2024-12-26:
Why are you reading `u16`, not `u8`?

saqibjaved977, 2024-12-27:
I have fixed the Rx and changed configuration to read u16. I have also added block!(rx.read()).unwrap() and block!(tx.write(received).unwrap() . 

I am able to receive the what I send from the phone. But the Tx is not working, its not transmitted back.